TomMajor/SmartHome #2: HB-UNI-Sensor1: Batteriespannung messen (ADC)

codmpm, 2018-05-10:
Hallo Tom,

vielen vielen Dank, dass du dir die Mühe mit den Beispielen gemacht hast. Ich habe das Ganze mal aufgebaut und erfolgreich mit Homegear gepaired. Ich überlege außerdem eine kleine Platine dafür zu bauen... das Universalsensor Projekt im FHEM-Forum ist leider eingeschlafen.

[image]

Nun würde ich gerne den Batteriezustand zusätzlich abfragen. Wenn ich es richtig verstanden habe, wird dazu ein Spannungsteiler benötigt der an einem analogen Pin des ATMega hängt. Oder sollte besser die Schaltung mit dem Mosfet aus HB-SEC-WDS-2 genommen werden?

Kannst du vielleicht kurz ein Beispiel geben, wie ich im Code `batterySensor` entsprechend parametrieren kann?

Freue mich über deine Antwort.

Gruß,
Patrik

TomMajor, 2018-05-10:
Hallo Patrick,
netter Testaufbau.

Du kannst normalerweise die Spannung intern ohne Spannungsteiler messen lassen, das machen die meisten sketche bei der AskSinPP Lib. Damit wird BatLow Zustand automatisch an die Zentrale übertragen wenn er eintritt. Im sketch macht das diese Zeile
`battery.low(22);`

Batteriespannung als Wert zusätzlich zu übertragen steht noch auf ToDo in meinem sketch.
Wenn Du noch bis morgen wartest baue ich das auch in mein Universalsensor Bsp ein.

Weiterhin steht auf meinem Plan, beim Wassermelder die Spannung ext. mit Spannungsteiler und Mosfet zu messen. Das ist schon in der HW vorbereitet, ich muss es nur noch testen bevor ich es commite. Sollte in den nächsten Tagen passieren.
MAX1724 braucht man mit einer Zelle. Und dann ext. Spannungsteiler da vor dem MAX1724 gemessen werden muss. Das alles wird mein Wassermelder Bsp. machen.

codmpm, 2018-05-11:
Hey Tom,

super - ich dachte mir schon sowas. Ich plane den Betrieb an einer 18650er Zelle, würde also `battery.low` auf `27` setzen. Die genaue Batteriespannung ist auf jeden Fall interessant, auch wenn es ja erst mal reicht zu wissen, wann die Batterie zu neige geht.
Vielleicht kannst du die Werte in eine Config-Variable auslagern? Ansonste kann ich das Projekt auch forken und einen PR machen.

Aktuell nutze ich den Arduino ProMini Bootloader bzw. Optiboot. Sollte ich BOD (Brown-Out) deaktieren?

Gruß,
Patrik

TomMajor, 2018-05-11:
Habe den Batt.messwert zur Übertragung impl., muss er nur später noch testen.
lowBatt Wert ist bereits konfigurierbar.
BOD kostet ca. 18uA mehr sleep Strom, je nachdem was Du erreichen willst.

TomMajor, 2018-05-11:
Habe in HB-UNI-Sensor1 die Übertragung der Batteriespannung getestet und commited.
lowBatt Wert ist wie gesagt über die Zentrale (CCU/RaspberryMatic/FHEM) konfigurierbar.
Der Kommentar
`// see my HB-SEC-WDS-2 for an example with higher resolution`
bezieht sich auf die Messung mit ext. Spannungsteiler und Mosfet und wird in ca. 2 Tagen im HB-SEC-WDS-2 Bsp. online sein.

codmpm, 2018-05-12:
Funktioniert, super!

War es richtig die Batteriespannungen hier zu konfigurieren? 
https://github.com/TomMajor/AskSinPP_Examples/blob/master/HB-UNI-Sensor1/HB-UNI-Sensor1.ino#L79

TomMajor, 2018-05-13:
Ja, genau dort. Mach das gleiche sicherheitshalber noch mal in Zeile 162. Werde das noch ändern über ein define, das der Wert nur einmal def. werden muss.
Und bei Zeile 80 den kritischen Wert. Ab diesen Wert macht das device nichts mehr sondern geht in deep sleep um eine Tiefentladung zu verhindern.

TomMajor, 2018-05-25:
Übertragung der Batteriespannung getestet und commited.

codmpm, 2018-08-12:
Hey Tom,

[image]

irgendwie komme ich gerade nicht weiter was die Laufzeit des Sensors angeht. Meine 18650er Zelle hat 2300mAh und der Sensor hält ca. 2,5 Monate. Sollte es nicht massiv länger sein? 

Muss ich noch etwas beachten, damit die Battereispannung richtig gemessen wird (blaue Linie)? 
Zum Vergleich die gelbe Linie, die einen ESP8266 (Wlan) mit 15min Sleep an 18650er Zelle darstellt.
Habe ich etwas missverstanden bezüglich der längeren Laufzeit des ATmegas und der genutzten low-power lib?

Gruß,
Patrik

TomMajor, 2018-08-12:
Ja, diese Zelle sollte def. länger halten.
Welches Sendeintervall hast du eingestellt?
Bei Problemen dieser Art wirst du nicht umhin kommen, die Stromaufnahme im Normalbetrieb und im Sleep zu messen.
siehe hier
https://github.com/TomMajor/AskSinPP_Examples/tree/master/Info/SleepTest
Wenn du große Abweichungen zu den 3mA bzw. 4uA hast würde das das Problem erklären..

codmpm, 2018-08-12:
Werd ich mal testen...

Das Sendeinterval habe ich mit Wissen nicht geändert, sollte also bei `300` default liegen.

closed 2018-11-09 by TomMajor