TomMajor/SmartHome #3: Universal Sensor mit E-Paper

knopserl, 2018-05-10:
Der Universalsensor ist genau das was ich suche/brauche. Allerdings möchte ich für die ensor an die Wand montieren und mit einer E-Paper Anzeige ausstatten.
Da der ATMEGA328p (Pro Mini) Flash dann sofort überschritten wird, bin ich auf den STM32F1 mit 128kB umgestiegen. auch das ist der Flash dann schnell auf 93% gestiegen und das RAM ist so gut wie voll. Die SPI E-Paper brauchen 20-30kB RAM und damit ist auch der STM32 aus dem Rennen.
Die ESP (8266 oder 32) haben 4-16MB Flash und 520kB RAM. Das reicht in jedem Fall.
Batterietauglich sind sind mit dem ULP Prozessor auch. 
Mein Problem: Es gibt keinen ESP Support für die AskSinPP Library und Pa-Pa hat schon abgesagt, also hab ich es selbst versucht und bin schon fast durch alles Files durch und habe die notwendigen Anpassungen gemacht und die meisten Compiler-Fehler sind mal weg (was noch nichts sagt). Allerdings stehe ich bei der Alarmclock Class an. Dazu reichen meine Kenntnisse nicht. Die Frage ist, ob es jemand gibt, der mir hier helfen könnte. Die ESP's (zumindest der ESP32) hat an sich gute Timer und RTC Funktionen aber ich verstehe den Source der Alarmclock-Klasse nicht wirklich und müsste ohne Hilfe aufgeben.

codmpm, 2018-05-11:
Was spricht gegen einen Arduino Mega (ATMega2560)? Damit sollte AskSinPP sogar "out of the box" laufen.

Siehe https://github.com/pa-pa/AskSinPP/blob/master/BatterySensor.h#L16

TomMajor, 2018-05-11:
ich glaube die 256k Flash reichen ihm auch nicht..
meine Meinung zum Thema habe ich gestern hier gepostet:

> Why not use the ESP32 for your demanding e-paper application, use the AVR for Bidcos and connect them via SPI or any other interface that suits your needs? Guess that would be much less work than porting AskSinPP to ESP32. Just my opinion :)

https://github.com/pa-pa/AskSinPP/issues/26

Aus meiner Sicht der wenigste Aufwand, für einen Port von papas AskSinPP von AVR auf ESP32 muss man sich warm anziehen und gutes HW know-how beider CPUs haben..

codmpm, 2018-05-11:
Das ist natürlich völlig richtig. Kommunikation würde sogar über (Soft-)Serial funktionieren. Je nachdem ob dafür auf dem 328p noch genug platz ist. Der ESP macht das ohne Probleme und da ja beide sowieso auf 3,3V laufen...

knopserl, 2018-05-11:
der ATMega2560 hat 8kB RAM, eine E-Paper Firmware braucht gut 32kB RAM. Das bieten die ATMEGAS nicht und der STM32 hat 20kB. Damit ist die Sache klar. Flash hätte mit 128kB im Notfall gereicht, 256kB auf jeden Fall.
Damit ist klar, dass die ESP's (vor allem der ESP32) die beste Option sind. Under das ESP32 Modul WROOM32 kostet 3,50€ und hat 16MB Flash und 520kB RAM.

Ich habe mich schon auf einigen Aufwand eingestellt und einen Großteil der Files schon umgestellt, das ja das meiste MCU unabhängig ist.
Wo ich schon kämpfe ist der Teil mit AlarClock und Systemclock, obwohl der ESP32 viel mehr timer und RTC Features schon on-board hat im Vergleich zu den ATMEGAS und den STM32's.

Im schlimmsten Fall muss ich halt auf das MySensors Projekt umsteigen, dort ist der ESp32, RFM95 (LoRa) schon unterstützt, aber damit sind meine Homematic Vorarbeiten (MapleCUN etc.) fast umsonst (außer das eini gekaufte Gerät).

Es haben bis jetzt halt wenige Leute einen Wand-Thermostat oder wenigstnes ein präzises Wand-Klimagerät mit E-Paper für Batterie gebaut. Sonst hätte ich auf das aufgesetzt.

codmpm, 2018-05-11:
Ach RAM, sorry... bin von Flash ausgegangen. Mein MCU-C ist leider nicht so gut, als dass ich dir dabei helfen könnte.

TomMajor, 2018-05-11:
> der ATMega2560 hat 8kB RAM, eine E-Paper Firmware braucht gut 32kB RAM. Das bieten die ATMEGAS nicht und der STM32 hat 20kB. Damit ist die Sache klar. Flash hätte mit 128kB im Notfall gereicht, 256kB auf jeden Fall.

ich habe früher mit dem STM32F407 gearbeitet, der hat 0,5-1MB Flash und 192kB RAM. Eventuell wäre das eine Alternative, weiß aber nicht wie einfach Arduino auf dem zum Laufen kommen würde.

> Ich habe mich schon auf einigen Aufwand eingestellt und einen Großteil der Files schon umgestellt, das ja das meiste MCU unabhängig ist.
Wo ich schon kämpfe ist der Teil mit AlarClock und Systemclock, obwohl der ESP32 viel mehr timer und RTC Features schon on-board hat im Vergleich zu den ATMEGAS und den STM32's.

Die ganzen generischen Klassen aus der AskSinPP umzustellen ist sicher nicht das Problem. Bei den HW-abhängigen Codestellen geht es dann halt ans Eingemachte. Deswegen meinte ich man braucht gute HW Kenntnisse von beiden CPUs um den Port zu machen, letztendlich muss man an jeder Stelle wo auf AVR Peripherals zugegriffen wird, genau verstehen was da geschieht um es auf die andere CPU zu portieren. Alles machbar, wird halt nur viel Zeit kosten..

jp112sdl, 2018-05-13:
@knopserl 
> der ATMega2560 hat 8kB RAM, eine E-Paper Firmware braucht gut 32kB RAM.

Ich betreibe [dieses ePaper Display](https://www.embeddedartists.com/products/displays/lcd_27_epaper.php) seit einiger Zeit an einem Mega2560 und lasse Grafiken, die von einer microSD geladen werden, anzeigen.
RAM ist zwar knapp, aber ausreichend.

knopserl, 2018-05-14:
Auf der AskSinPP project Seite habe ich den Stand meiens ESP32 Ports beschrieben. Pairing und senden von Werten funktioniert schon. Was noch nicht geht sind die RTC und Action Klasse. Damit der Sleep/Wake Modus in der Loop Section des onsor-Programmes.
Das geht auch übrigens noch nicht im STM32 Port. Wurde also für diese MCU auch noch nicht gelöst (und ich habs auch für den ESP32 noch nicht portieren können). Was mir u.a. helfen würde, wäre eine LowPower Lib für ESP32 (der kann ja einige Sleep/Hibernate Modes und auch eine RTC). Aber dazu muss ich diese Source Teile von Pa-Pa noch besser verstehen lernen.

TomMajor, 2018-05-25:
Hauptthread für ESP support ist hier:
https://github.com/pa-pa/AskSinPP/issues/26

closed 2018-05-25 by TomMajor