TomMajor/SmartHome #4: TSL2561 - plausible Werte

jp112sdl, 2018-06-08:
Hallo TomMajor,

dem Kommentarblock entnehme ich ähnliche Werte, wie ich sie bei mir gemessen hatte.

https://github.com/TomMajor/AskSinPP_Examples/blob/b81944e1d8883367090fad9beb27f845727a4837/HB-UNI-Sensor1/Sensors/Sens_Tsl2561.h#L119

Beziehen sich diese Kommentare auf Messungen vor den letzten Änderungen im Code?
Werden jetzt auch bei direktem Sonnenlicht plausible Werte angezeigt?
Habe derzeit leider keine Möglichkeit, es selbst zu testen.

LG,
JP

TomMajor, 2018-06-08:
Hallo JP,
die Kommentare geben die (Roh)-Daten wieder, so wie sie vom Sensor (bzw. aus der Adafruit Lib) bei verschiedenen Helligkeiten kommen, völlig egal wie der HB-UNI-Sensor1 code dahinter aussieht.

Meine Codeänderung vor 2 Tagen setzt automatisch die Empfindlichkeit runter, falls der Sensor übersteuert.

Falls auch die unempfindlichste Stufe nicht ausreicht wird ein Wert von 130000 Lux (klarer Himmel und Sonne) an das SmartHome System geliefert um plausibel zu bleiben und helligkeitsgesteuerte Aktionen nicht zu verderben :)
z.B. wenn der Sensor direkt in Richtung Sonne blickt, da kommt kein vernünftiger Messwert, aber der overflow wird erkannt.

jp112sdl, 2018-06-09:
Ich habe mir jetzt mal einen Sketch+Addon+Hardware gebaut, der einen TSL2561 und einen BH1750 beinhaltet, die Werte alle 60 Sek. sendet und diese in ein Diagramm schreibt.
Mal schauen, wie sich beide Sensoren im Alltag bewähren.
<hr>
Btw.: Der TSL2561 kann max 65535, so dass ein uint16_t ausreicht und man die Payload um 2 Byte kürzen könnte!?

Der BH1750 ist auf denselben Max-Wert begrenzt.
<hr>
Im originalen HM-Sen-LI-O wird vermutlich ein MAX44009 verbaut sein. 
Der kann bis 188.000 Lux. 
Ich werde mal einen bestellen und auch mit diesem ein paar Testreihen durchführen.

jp112sdl, 2018-06-09:
Zwischenstand:
Bei absoluter Dunkelheit sendet 
- BH1750: 0 Lux
- TSL2561: 130000 Lux.

Da ist noch irgendwo ein Bug in der Sensorklasse

[image]

TomMajor, 2018-06-09:
> Da ist noch irgendwo ein Bug in der Sensorklasse

Bei mir ging der Kurztest, konnte aber die neue Sensivity Automatik noch nicht testen, mache ich morgen.

> Btw.: Der TSL2561 kann max 65535, so dass ein uint16_t ausreicht und man die Payload um 2 Byte kürzen könnte!?

Habe mich an die Vorlage im HM xml gehalten von deren Wetterstation. Lux können ja grösser 65k sein, deswegen die 4 Byte. raw value beim TSL2561 != Lux. Ausserdem gibt es 2 raw values (full & IR spectrum).

jp112sdl, 2018-06-09:
> Habe mich an die Vorlage im HM xml gehalten von deren Wetterstation.

Die echte HM-Wetterstation HM-WDS100-C6-O-2 liefert nur Helligkeitswerte von 0-255.
https://github.com/pa-pa/AskSinPP/blob/master/examples/HM-WDS100-C6-O-2/HM-WDS100-C6-O-2.ino

Aber ich denke mal, du meinst den Helligkeitssensor HM-Sen-LI-O, der 0 - 200.000 Lux kann.

> Lux können ja grösser 65k sein, deswegen die 4 Byte. raw value beim TSL2561 != Lux. Ausserdem gibt es 2 raw values (full & IR spectrum).

Ah ok, dann hab ich mich da geirrt.

jp112sdl, 2018-06-10:
Kurzer Zwischenstand von gestern Abend bis heute 09:15 Uhr.
grüne Kurve = BH1750
rote Kurve = TSL2561

[image]

TomMajor, 2018-06-12:
interessante Kurven.

Wenn der TSL2561 buggy sein sollte, wären die Rohwerte bei hohen Helligkeiten interessant, also das was bei getFullLuminosity() als 32bit Wert zurückkommt, da könnte man vielleicht mehr sehen wo es klemmt.
Damit wird der Lux Wert berechnet. Theoretisch könnte die Adafruit Lib bei der Lux Berechnung auch noch einen Bug haben.

Am besten wäre natürlich der Vergleich mit einem genauen Helligkeitsmesser.
Ich habe noch einen Lux Sensor hier liegen (GA1A1S202WP) den könnte ich mal parallel zum TSL2561 messen lassen, das wird aber nicht vor dem WE.

BH1750:

> Small measurement variation (+/- 20%)

Das sind auch Witzbolde bei Rohm :)

jp112sdl, 2018-06-13:
Bei uns ist es seit 2 Tagen stark bewölkt -.-

Sobald sich die Sonne wieder blicken lässt, logge ich `getFullLuminosity()` mal mit, falls du es dann nicht inzwischen bereits selbst gemacht haben solltest.

Nebenbei habe ich noch das hier gefunden - aber auch ohne Lösung:
http://community.mydevices.com/t/tsl2561-calculations-wrong-during-very-bright-light/6378

Alternativ werde ich dann auch mal die Sparkfun Lib testen.
https://github.com/sparkfun/SparkFun_TSL2561_Arduino_Library

TomMajor, 2018-06-13:
Genau, mit dem Wert aus getFullLuminosity() könnte man noch mal die Formel aus den Datenblatt nachrechnen, die Luxberechnung verwendet für versch. Teiler aus Vis/IR auch verschiedene Faktoren, ausserdem geht noch die Gehäuseausführung ein, wäre möglich dass für bestimmte Lichtbereiche noch ein Fehler drin ist.
Was def. nicht mit dem TSL2561 geht, das ist auch in Sensors/Sens_Tsl2561.h zu sehen:
Den Sensor direkt in die Sonne *ausrichten*. Dann kann man die Werte vergessen, sollte man bei der Platzierung des Gehäuses beachten.

jp112sdl, 2018-06-14:
Vielleicht hilft uns/dir das hier noch weiter?
http://www.netzmafia.de/skripten/hardware/RasPi/Projekt-Licht/index.html

TomMajor, 2018-07-06:
Heute die MAX44009 sensor class eingecheckt, dieser macht bei hohen Helligkeiten einen besseren Eindruck.
siehe [https://github.com/TomMajor/AskSinPP_Examples/tree/master/Info/SensorTest_Lux](url)
Der MAX44009 wird in den nächsten Tagen als Option im HB-UNI-Sensor1 erscheinen.

closed 2018-07-06 by TomMajor