TomMajor/SmartHome #5: Fehler im CCU2 Addon

WernerSch, 2018-11-08:
Hallo

Ich hoffe ich bin hier richtig
Das update vom CCU2 Addon scheint einen Fehler zu haben. In iobroker sehe ich plausible Luxdaten aber keine Batteriespannung mehr. In der CCU2 sehe ich keine Helligkeit aber die Batteriespannung die bei einem Sensor aber nicht stimmt.
Ist das Problem schon bekannt?

Gruß
Werner

WernerSch, 2018-11-10:
Hallo Tom

habe ich gemacht. In der CCU2 wird jetzt die richtige Spannung angezigt. In iobroker sehe ich aber immer noch keine Batteriespannung. Das ging die ganze Zeit.

Der Digitale Eingang ist für die Messung der Batteriespannung?

Gruß
Werner

TomMajor, 2018-11-10:
Nein, der dig. Eingang ist für das Einlesen eines 0/1 Zustandes, Schalter offen/geschlossen etc.

Also wenn in der CCU alles richtig zu sehen ist glaub ich nicht dass es am sketch oder addon liegen kann, eventuell muss der iobroker irgendwie upgedatet werden?

Kenne mich (momentan) noch nicht mit iobroker aus, habe aber vor den aufzusetzen. Hast Du eventuell einen link auf eine Beschreibung wie man die CCU mit iobroker verknüpft?

Eventuell kann @jp112sdl was zu diesem iobroker refresh Problem sagen, da ich weiß das er ihn verwendet.

jp112sdl, 2018-11-10:
@WernerSch 
Hi!
Hast du die hm-rpc-Instanz mal neu synchronisiert?
Einstellungen der Instanz-> Haken setzen bei "Synchronisiere Geräte neu (einmalig)"

Oftmals hilft es auch, den Ordner des Gerätetyps in Objekte->hm-rpc.meta->VALUES zu löschen und anschließend die Instanz neu zu starten

WernerSch, 2018-11-10:
@jp:  das scheint geholfen zu haben. Nur heisst es jetzt OPERATING_VOLTAGE und nicht mehr Batteriespannung Den datenpunkt Batteriespannung habe ich mal gelöscht.  

@tom: ich gehe mal davon aus das hm-rpc.0.UNISENS002.1.ILLUMINATION der Digitale Port ist. 
In der CCU2 habe ich den Kanal zu den Favoriten. Da habe ich jetzt 2 mal das Feld Helligkeit. Eins zeigt den Sensor an das andere ist bei beiden Sensoren 0 (No_Sensor). Ist das der Port bzw. stimmt bei mir in der CCU2 auch noch was nicht?

Gruß
Werner

jp112sdl, 2018-11-10:
@WernerSch 
Hast du nach der Installation des neuen Addons und dem Flashen mit der neuesten Sketch-Version das Gerät in der CCU auch entfernt und neu angelernt?

Das ist zwingend erforderlich, wenn sich Datenpunkte ändern.

@TomMajor 
Normalerweise legt man eine neue XML- und Sketch-Firmware-Version an.
Schau dir mal die `hb-uni-sen-wea*` Dateien in meinem [Addon](https://github.com/jp112sdl/JP-HB-Devices-addon/tree/master/src/addon/firmware/rftypes) an.
Bei der Wetterstation hatte ich auch verschiedene Entwicklungsstadien.
Einige Nachbauer haben schon eine frühe Version im Einsatz, die ihnen völlig ausreicht.
Das Problem ist nämlich nun: 
Wenn jemand das neue Addon installiert bei dem du die bestehende XML einfach austauschst, funktionieren bereits laufende HB-UNI-Sensor Geräte nicht mehr - man muss sie also alle neu flashen, neu anlernen und in Programmen ggf. wieder hinzufügen.

TomMajor, 2018-11-10:
@WernerSch 
Hast du das aktuelle Addon 1.55 installiert?
ILLUMINATION wird dort gepatched und sollte als "Digitaler Eingang" erscheinen, siehe meine Bilder in der Readme des Unisensors.
Details zum Patch:
https://github.com/TomMajor/AskSinPP_Examples/blob/master/HB-UNI-Sensor1/CCU_RM/README
(ganz unten)
Kannst du mal checken ob die dort beschiebenen Änderungen in den 3 Dateien bei dir erscheinen oder nicht?

TomMajor, 2018-11-10:
@jp112sdl

Also die VERSION habe ich in der Addon source natürlich hochgesetzt.
https://github.com/TomMajor/AskSinPP_Examples/blob/master/HB-UNI-Sensor1/CCU_RM/src/addon/VERSION

Muss ich in der xml auch eine Version hochsetzen?
also z.B. das hier aus einem deiner xml?
`<device version="1" rx_modes="BURST" supports_aes="true" cyclic_timeout="88200">`
Dort die 1 hochzählen?
Dann ist man sauber?

jp112sdl, 2018-11-11:
@TomMajor 
> Also die VERSION habe ich in der Addon source natürlich hochgesetzt.

Diese Versionsangabe dient lediglich der Anzeige in der WebUI, um den Anwender über den Versionsstand des jeweiligen Addons zu informieren.

===

Was ich meine, ist die Firmware-Version des Gerätes.

Du hast zum einen die FW-Versionsangabe im Sketch https://github.com/TomMajor/AskSinPP_Examples/blob/ed370ff514eba719e46ae32fde048a8f0348cf0e/HB-UNI-Sensor1/HB-UNI-Sensor1.ino#L96
In dem Fall `0x10` was Version "1.0" entspricht und in der WebUI in den Geräteeinstellungen auch "Version 1.0" anzeigen lässt.
Die FW Version wird beim Anlernen mit übertragen.

Auf der Gegenseite, der CCU, liegt die XML.
Dort ist die Firmware-Version, für die diese XML gültig ist, beschrieben in https://github.com/TomMajor/AskSinPP_Examples/blob/ed370ff514eba719e46ae32fde048a8f0348cf0e/HB-UNI-Sensor1/CCU_RM/src/addon/firmware/rftypes/hb-uni-sensor1.xml#L5

Diese Gerätebeschreibungsdatei gilt für Geräte mit einem FW-Stand GE (>=) 12(dec).
Da 10(hex) größer als 12(dec) ist,  wird diese XML beim Anlernen gezogen.

Es wäre nun also ratsam
 - die Vorgängerversion der XML auf die FW-Version `0x10` festzunageln (`cond_op="E" const_value="0x10"`) und umzubenennen (z.B. `hb-uni-sensor1_e_0x10.xml`)   
- die `//Firmware Version` im aktuellen Sketch zu ändern, z.B. auf `0x11` und die XML genau an diese FW-Version zu binden mit `cond_op="E" const_value="0x11"`

Anschließend beide Versionen ausliefern.

So können dann nach wie vor auch Geräte mit altem FW-Stand wieder korrekt angelernt werden, ohne sie neu flashen zu müssen.

WernerSch, 2018-11-11:
Hallo

oh das ist jetzt aber jede Menge Info. ich werde das am besten Schritt für Schritt machen und mich melden.

Ab- und Anmelden habe ich zweimal gemacht. Hat sich nichts geändert. In der CCU2 2 mal Helligkeit und in iobroker (daten waren weg) wieder illumination.

Gruß
Werner

WernerSch, 2018-11-11:
Details zum Patch:
https://github.com/TomMajor/AskSinPP_Examples/blob/master/HB-UNI-Sensor1/CCU_RM/README
(ganz unten)
Kannst du mal checken ob die dort beschiebenen Änderungen in den 3 Dateien bei dir erscheinen oder nicht?

Da habe ich das problem das die Datei schreibgeschützt ist und ich mit chmod den Fehler read-only filesystem bekomme. Habe momentan nicht mehr im Kopf wie das geht. Will die CCU2 ja nicht killen. Da muss ich erst mal nachschlagen wie das geht. Das wird erst später was. Da gehts dann weiter

Gruß
Werner

WernerSch, 2018-11-11:
Also mit 
mount -o remount,rw / 
ging es dann. Habe die zwei Zeilen hinzugefügt.

Die Zeile
Ersetzen:
/www/webui/js/lang/de/translate.lang.diagram.js
"diagramValueTypeILLUMINATION": "Digitaler Eingang",
stand so schon drin, da habe ich nichts gemacht. Werde die CCU jetzt mal rebooten

Gruß
Werner

TomMajor, 2018-11-11:
@jp112sdl 
Danke für die ausführliche Erklärungen zur Versionierung Firmware/xml.
Werde ich heute noch einpflegen für die aktuelle Version.
Für die Vorgängerversion möchte ich das eher nicht machen, ich müsste ja dann auch das Addon seperat dafür anbieten und eine komplette "Versionspalette" supporten, das kann ich nicht leisten. Letztendlich ist das hier ein Demo und Anregung zum Selbstbau. Ich werde mich einfach bemühen die Payload nicht mehr (so bald) zu ändern :)

TomMajor, 2018-11-11:
@WernerSch 
Willst du damit sagen, die Änderungen waren in
/www/webui/js/lang/de/translate.lang.diagram.js
drin
aber nicht in
/www/config/stringtable_de.txt
/www/webui/js/lang/de/translate.lang.stringtable.js

Bist du sicher? Dann hätte das Addon install noch einen Fehler für die CCU2.
Ich kann nur auf der RaspberryMatic testen und da lief es.

jp112sdl, 2018-11-11:
>  ich müsste ja dann auch das Addon seperat dafür anbieten und eine komplette "Versionspalette" supporten, 

Nein eben nicht. Du hast ein Addon (Anzahl 1) und packst da einfach mehrere XML-Files mit rein.

Ich habe auch nur ein Addon, welches 4 oder 5 Versionen der Wetterstation unterstützt.

jp112sdl, 2018-11-11:
@WernerSch 
> Werde die CCU jetzt mal rebooten

Unbedingt den **Browsercache** nach solchen Modifikationen löschen!

Bei Anpassungen an Texten etc., die die WebUI betreffen, arbeite ich grundsätzlich im Inkognito Modus des Browsers, um garantiert keine gecachten Inhalten zu sehen.

WernerSch, 2018-11-12:
Hallo

@TomMajor 
also ich habe recht genau hingeschaut. Ich hätte vielleicht besser die Dateien gespeichert zur Dokumentation. Das werde ich nächstes mal machen.

@jp112sdl 
das mit dem Inkognito Modus werde ich heute Abend mal versuchen

Gruß
Werner

WernerSch, 2018-11-12:
Hallo

also der Privatemodus in Firefox hat auch nichts gebracht.

gruß
Werner

TomMajor, 2018-11-12:
Hallo Werner,

ich habe gestern Nacht noch mal den "Lock" in der Version zwischen Firmware und xml commited.

Bitte das machen:

- den vorhandenen Sensor aus der Zentrale löschen
- das vorhandenen Addon deinstallieren
- browser cache löschen
- den Sensor mit dem aktuellen sketch flashen
- das aktuelle Addon installieren
- Sensor anlernen
- hm-rpc-Instanz neu synchronisieren: Einstellungen der Instanz-> Haken setzen bei "Synchronisiere Geräte neu (einmalig)"

Danach prüfe bitte mal ob die 3 Änderungen

1. neue Zeile in:
/www/config/stringtable_de.txt
WEATHER|ILLUMINATION\t\t\${stringTableWeatherIllumination$JAVA_DEVICE_NAME}

2. neue Zeile in:
/www/webui/js/lang/de/translate.lang.stringtable.js
"stringTableWeatherIllumination$JAVA_DEVICE_NAME" : "Digitaler Eingang",

3. Ersetzen in:
/www/webui/js/lang/de/translate.lang.diagram.js
"diagramValueTypeILLUMINATION": "Digitaler Eingang",

auch so in den 3 Dateien angekommen sind. Nur für das Prüfen kannst du den read-only Modus der CCU lassen.

Gestern noch mal auf der RaspberryMAtic getestet, läuft einwandfrei.
Wenn wir für die CCU2 leichte Modifikationen brauchen mache ich das gern, ich muss nur wissen wo die Abweichung liegt..

kpwg, 2018-11-13:
Das Problem betrifft auch FHEM: durch die geänderte Payload habe ich bei meinem Produktivsystem die alte Variante ohne den Kontakt beibehalten. Ich müsste nun bei vier Sensoren auf Basis papa-2AAA-SHT10 zum flashen nochmal ent- und wieder verlöten, was ich sehr gerne vermeiden möchte. Besteht die Möglichkeit eines Versionsmanagementes, d.h. das die Sensoren ab Änderung des Payload eine neue Versionsnummer bekommen? Ich kann gerne nochmals Hardware beisteuern ;-)

Viele Grüße, Ricardo

WernerSch, 2018-11-13:
Hallo Tom

es hat sich leider nichts geändert. 
Ich habe die Dateien auf den Stick geladen und dann danach gesucht, das ist besser

Ist nicht vorhanden
    neue Zeile in:
    /www/config/stringtable_de.txt
    WEATHER|ILLUMINATION\t\t${stringTableWeatherIllumination$JAVA_DEVICE_NAME}

die ist vorhanden
    neue Zeile in:
    /www/webui/js/lang/de/translate.lang.stringtable.js
    "stringTableWeatherIllumination$JAVA_DEVICE_NAME" : "Digitaler Eingang",

da steht "diagramValueTypeILLUMINATION": "Digitaler Eingang", drin
    Ersetzen in:
    /www/webui/js/lang/de/translate.lang.diagram.js
    "diagramValueTypeILLUMINATION": "Digitaler Eingang",

Was mir aufgefallen ist. Ich habe den Sensor vorher auf der CCU2 gelöscht. Nach dem flashen wieder angelernt und in den Einstellungen war noch der Höhenparameter drin, der normal doch 0 ist beim anlernen.

gruß
Werner

jp112sdl, 2018-11-13:
@WernerSch 
> Nach dem flashen wieder angelernt und in den Einstellungen war noch der Höhenparameter drin, der normal doch 0 ist beim anlernen.

Die Werte werden im EEPROM abgelegt und beim Neu-Flashen nicht (zwingend) gelöscht.
Am saubersten ist da ein komepletter RESET (Anlerntaster mind. 6 Sekunden gedrückt halten) vor dem Neuanlernen.

@TomMajor 
Wenn du noch Hilfe wegen der Versionierung von Sketch + XML benötigst, dann schick mir eine PN im HM-Forum. Ich denke, du wirst da nicht dumherumkommen... sonst sind alle bisherigen HB-UniSensor Nutzer aufgeschmissen, sobald sie das neue Addon installieren.

TomMajor, 2018-11-13:
Hallo Leute, zunächst mal:
Ich habe diese Woche und auch kommendes WE sehr wenig Zeit und werde außerdem ein paar Tage unterwegs sein, kann also mal etwas länger dauern mit einer Antwort.

Versionierung werde ich bei der nächsten Änderung der payload definitiv machen, jetzt da ich einiges drüber gelernt habe. Wobei ich wie gesagt nicht vorhabe die payload so schnell noch mal zu ändern ;)

Sauber rückwirkend kann ich das nicht machen da z.B. für FHEM ja beide Versionen mit ID F103 daherkommen. Für HM gäbe es den Weg von Jerome mit dem zweiten xml, das habe ich verstanden und kann ich demnächst nachreichen.

@kpwg 
Bitte am Besten für FHEM eine neue issue aufmachen da es hier um das CCU2 addon geht.

@WernerSch 
Wenn ich dich richtig verstehe kommt die Änderung nur in der Datei
/www/config/stringtable_de.txt
nicht an?
@jp112sdl 
Hast du eine Idee warum das bei der CCU2 nicht geht? Irgendwelche Spezialitäten mit den Rechten dort?
Ich habe diese HW nicht.
Das ist der code:
`myfile="/www/config/stringtable_de.txt"`
`mysearch="#[[:space:]]*END WEATHER SENSOR"`
`myinsert="WEATHER|ILLUMINATION\t\t\${stringTableWeatherIllumination$JAVA_DEVICE_NAME}"`
`if [ -z "`cat $myfile | grep \"$JAVA_DEVICE_NAME\"`" ]; then`
`	sed -i "/$mysearch/i $myinsert" $myfile`
`fi `

Kann eventuell Werner mit ausreichend Linux Kenntnissen das auf der Kommandozeile debuggen was da schiefläuft?
Hat stringtable_de.txt andere Rechte als die anderen beiden Dateien?

Ist das mit dem "custom Namen" Digitaler Eingang jetzt das einzige Problem oder was geht außerdem nicht auf der CCU2?

jp112sdl, 2018-11-14:
Ich habe leider auch keine CCU2 mehr zum Testen, da diese nun als LAN Gateway läuft.
Grundsätzlich ist dort aber nichts anders als bei der CCU3 oder RaspberryMatic.
Teilweise halt ältere Binaries von `wget`, `sed` & co.

Das Einzige, was mir auf Anhieb auffällt, womit ich auch schon Trödel hatte, sind die Tabs.
`WEATHER|ILLUMINATION\t\t\${stringTableWeatherIllumination$JAVA_DEVICE_NAME}`
-> da sind 2 Tabs drin. Normalerweise wird immer nur 1 Tab verwendet.

Hätte evtl. auch erwähnt werden sollen, dass `$JAVA_DEVICE_NAME` durch das Wort `HBUNISensor1` ersetzt werden müsste, wenn man die Dateien manuell editiert!?
Ich weiß jetzt nicht ganz genau, ob das zu Unstimmigkeiten geführt haben könnte.

WernerSch, 2018-11-14:
Hallo Tom

ja, ich habe die Dateien extra auf den Stick kopiert damit ich besser suchen kann.

Das ist kein Problem wenn es etwas dauert, ich warte ab und schaue als mal rein

Gruß
Werner

TomMajor, 2018-11-15:
@jp112sdl 
Danke für deine Test und deine Hartnäckigkeit :1st_place_medal: 
Ja, die Tabs scheinen zickig zu sein. Ohne Tabs ging gar nichts bei mir auf RM als ich diesen code für stringtable_de.txt gemacht hatte, deswegen habe ich auch das in der readme notiert:

    wichtig, Tabs müssen sein, nur mit space wird der String-Namen nicht eingelesen!

Echter tab wäre noch eine Option zum Testen für Werner.
Ich schaue das ich das noch morgen oder Fr integrieren kann, leider aktuell wenig Zeit.

jp112sdl, 2018-11-17:
@TomMajor 
> wenn man das nächste Mal die Model Nummer hochzählt geht das auch für FHEM, hoffentlich macht das dann keine Probleme in HM/RM...

Kann FHEM nicht mit unterschiedlichen Versionsständen (index="9") desselben Gerätetyps umgehen?
Das gabs ja bei den echten HM-Geräten schon immer.

Natürlich kannst du auch mit unterschiedlichen Modell-IDs arbeiten. 
Ich weiß nur nicht, ob die in der XML auch alle dieselbe ID (`id="HB-UNI-Sensor1"`) haben dürfen. Vermutlich nicht. Um die Eindeutigkeit zu gewähren, müsstest du dann da evtl dann `-1`, `-2` usw. anhängen.

TomMajor, 2018-11-17:
@jp112sdl 
Habe mir das Perl script noch mal zu Gemüte geführt, ich glaube das bekomme ich hin ähnlich wie bei HM/RM auch für FHEM die Firmware version zu verwenden um Daten unterschiedlicher Sensor Versionen an FHEM zu pushen.

@kpwg 
Planänderung: Bitte behalte mindestens einen Sensor mit der alten Firmware. Ich mache die nächsten Tage eine FHEM script Version die mit beiden zurechtkommen sollte, ich sag per PM Bescheid.
jp112sdl hat mich wieder mal auf den richtigen track gebracht :)

WernerSch, 2018-11-19:
Hallo Tom

ich habe ein Update gemacht und die Änderung ist jetzt drin. Die Software am Sensor habe ich nicht geändert. In der CCU wird der Port aber immer noch als Helligkeit angezeigt.

gruß
Werner

TomMajor, 2018-11-19:
Hallo Werner,
die CCU2 ist aber ein Biest :)
Packe mir bitte mal diese 3 Dateien nach der Installation des AddOn zusammen und lade sie mir irgendwo hoch, dropbox etc. :
/www/config/stringtable_de.txt
/www/webui/js/lang/de/translate.lang.stringtable.js
/www/webui/js/lang/de/translate.lang.diagram.js
Du musst read-only dafür nicht ändern, einfach nur auf deinen lokalen PC kopieren und packen.

jp112sdl, 2018-11-20:
> die CCU2 ist aber ein Biest :)

Ich glaube da eher weniger an das Modell "CCU2" als das Problem. ;)
Vielmehr vermute ich, dass bei der früheren Installation schon eine Modifikation mit _falschem_ Tab in die Stringtable gekommen ist, der da nun vorhanden ist und auch vom Skript nicht "repariert" wird.

WernerSch, 2018-11-20:
Hallo Tom 

die sind ja klein, die kann ich auch per email schicken. Wie du es gerne hättest.
Brauche dann noch eine emailadresse

Gruß
Werner

TomMajor, 2018-11-20:
einfach einen upload nach dropbox o.ä. zum download für mich würde ich bevorzugen.

WernerSch, 2018-11-20:
und auf welche emailadresse?

Gruß
Werner

TomMajor, 2018-11-20:
Also wenn du den upload link hier nicht posten möchtest kannst du mir eine PM im HM Forum schicken.

TomMajor, 2018-11-25:
Zur Vollständigkeit der enthaltenen Infos hier auch das Resultat der PN Konversation:

jetzt ist der Digitale Eingang da
Kann ich noch was tun für dich oder ist das Thema erledigt?
WernerSch 

ok, danke für die Rückmeldung.
das Problem war definitiv das Tabzeichen \t im sed Kommando, die sed Version auf RaspberryMatic akzeptiert das, die alte sed Version auf CCU2 nicht. Mit einem "echten" Tab geht es in beiden Versionen.
@jp112sdl hatte wieder mal den Finger drauf :+1: 
Damit ist dieses Problem gelöst denke ich.

Als Anmerkung, ich werde dennoch den custom HM data type 'Digitaler Eingang', der mit dem Addon erzeugt wird, mit Wirkung der Addon Version 1.60 von heute wieder rausnehmen und dafür den nativen HM data type 'VALVE_STATE' (WebUI: Ventilposition) hernehmen.
Grund: Ich konnte zwar den Fake für die Zentrale machen, angeschlossene Systeme wie ioBroker lassen sich davon nicht beeindrucken und dies führt def. zu Verwirrung.
Deswegen wird jetzt VALVE_STATE für den Zustand des Eingangs herhalten.
Tom

jp112sdl, 2018-11-26:
Und was ist mit den Nutzern der CCU, die sich tatsächlich den `VALVE_STATE` in der WebUI anzeigen lassen? Das ist mit der neuesten RaspberryMatic sogar jetzt nativ mit eingebaut.

Desweiteren würde auch die Bezeichnung in Programmen dann verfälscht werden.
WENN "Ventilposition ..."

> Grund: Ich konnte zwar den Fake für die Zentrale machen, angeschlossene Systeme wie ioBroker lassen sich davon nicht beeindrucken und dies führt def. zu Verwirrung.

ioBroker wird sich auch bei `VALVE_STATE` nicht davon beeindrucken lassen, da externe Systeme nicht auf die CCU-internen Übersetzungstabellen zugreifen.

TomMajor, 2018-11-26:
@jp112sdl 
verstehe nicht ganz was du mir sagen willst.
Ich nehme einfach einen nativen Datentyp ohne die Translation dafür zu faken da ein Translation fake in ioBroker nicht ankommt.
Irgendeinen Datentyp muss ich ja für den digitalen Eingang nehmen, VALVE_STATE ist m.E. einer der wenigen die 'einigermaßen' zu einem digitalen Zustand passt.
Und die Bezeichnung in ioBroker soll einfach konsistent zu der gewohnten im WebUI sein, dass ist der Grund warum ich ab V 1.60 auf den Translation fake verzichten möchte.

TomMajor, 2018-11-27:
Ok, mit der neuen Firmware 1.2 läuft jetzt alles auch im ioBroker, der Zustand des digitalen Eingangs als Datentyp VALVE_STATE kommt an.
Problem gelöst, Danke an alle Beteiligten für die Mithilfe. :+1:

closed 2018-11-27 by TomMajor