TomMajor/SmartHome #6: CR: Vervollständigung Grafik

jp112sdl, 2018-12-12:
Hi,

um den Stromverbrauch noch weiter zu reduzieren, ist es evtl. auch ratsam die [die LED BUILTIN zu entfernen](https://jp112sdl.github.io/AskSinPPCollection/Grundlagen/01_hardware.html#batteriebetrieb)?

TomMajor, 2018-12-12:
prinzipiell ja, alles was leuchten kann entfernen :)
Die besagte LED ist am SCK und wird im AskSinPP Normalbetrieb nicht leuchten, da SCK mit 4 oder 8 MHz nur kleinste Zeiteinheiten high ist, wenn der CC1101 angesteuert wird, bin nicht sicher ob man den Spareffekt merken wird.
Wenn ich meine Strommessung beim Senden mache schaue ich mir auch mal SCK an.

jp112sdl, 2018-12-12:
Ich seh es halt weniger wissenschaftlich.

Benötigt man die LED? -> Nein, also weg damit.
Spätestens bei Geräten mit aktiviertem BurstDetector blinkert die LED alle paar Millisekunden vor sich hin _(nach derzeitigem Code-Stand)_, da lohnt es sich in jedem Fall.

Und auch wenn sich beim Sensor die Laufzeit nur um 1 Minute aufs Jahr gerechnet verlängert - das Entfernen der LED dauert ne Sekunde, wenn man eh schon dabei ist, die andere LED + LDO zu entfernen ;)

TomMajor, 2018-12-12:
Ganz ehrlich, bei 1 Minute Verlängerung der Laufzeit würde ich es Leuten nicht empfehlen, ist schließlich ein destruktiver Eingriff mit dem Potential den Pro Mini zu beschädigen - bei ungeübten Leuten. (bei der anderen LED und LDO natürlich auch, aber die sind wirklich notwendig).

Du siehst die SCK LED mit BurstDetector tatsächlich leuchten? Das wäre natürlich ein Grund.

jp112sdl, 2018-12-12:
> Du siehst die SCK LED mit BurstDetector tatsächlich leuchten? 

Korrekt. Alle [250ms](
https://github.com/pa-pa/AskSinPP/blob/master/Activity.h#L207) wird ein [`detectBurst`](https://github.com/pa-pa/AskSinPP/blob/master/Radio.h#L679) ausgeführt.

TomMajor, 2018-12-12:
Habe die Beschreibung entsprechend upgedatet.
@jp112sdl BTW, LED on/off ist jetzt einstellbar.

closed 2018-12-12 by TomMajor