TomMajor/SmartHome #7: Addon: "Herunterladen"-Button funktioniert nicht

jp112sdl, 2018-12-30:
**Fehler:**
Beim Klick auf "Herunterladen":
[image]
wird die `update-check.cgi` heruntergeladen _und nicht die `HB-UNI-Sensor1-addon.tgz`_.

Ich habe das vor kurzem bei meinem Addon auch [gefixt](https://github.com/jp112sdl/JP-HB-Devices-addon/commit/3774dfdb90675b0f00f914d5bdb7b553cee15ddc).

TomMajor, 2018-12-30:
Danke, schaue ich mir heute später noch an.

TomMajor, 2018-12-30:
@jp112sdl 
Bei mir funktioniert das Herunterladen einwandfrei.
Glaubst du es hat was mit der neusten RM Version zu tun? Weil ich bin auf 20181126 und damit eine hinter der aktuellen.

jp112sdl, 2018-12-30:
Ja ich glaube schon. 

Bei mir ging es zumindest nicht mehr und bei deinem Addon auch nicht mehr. Habe Chrome und Safari getestet. 

Ich wurde darauf auch nur von einem aufmerksamen Anwender hingewiesen. 

Mit den paar Zeilen Änderung ist alles wieder iO.