TomMajor/SmartHome #8: Neues Addon zeigt nichts an

WernerSch, 2018-12-30:
Hallo Tom

ich habe leider wieder ein Problem. Ich habe die neueste Firmware auf meinen Unisensor gespielt (BME280/MAX4409) und das Addon in der CCU2 aktualisiert. Den Sensor kann ich anlernen aber nicht einstellen. Die Seite bleibt leer. Ich kann auch keinen Kanal in die Favoriten aufnehmen, weil der Sensor dort in der Auswahl gar nicht erscheint.
Wo liegt das vorhergehende Addon? Dann würde ich es damit mal versuchen
Poweron Reset hat auch nicht geholfen

Gruß
Werner

TomMajor, 2018-12-30:
Hallo Werner,

ich habe seit Wochen nichts (Entscheidendes) am Addon geändert.
Es werden 2 alte Addons mitinstalliert damit alte Geräte mit alten Firmwares kompatibel bleiben, diese enden mit _fw0x11.xml bzw _fw0x10.xml im rftypes Ordner.
Prüfe bitte neueste Firmware geflasht, neustes Addon 1.63 installiert, Gerät in Zentrale löschen, Config Taster lange drücken für Geräte reset, dann neu anlernen. Mehr kann ich dazu momentan nicht sagen.

WernerSch, 2018-12-31:
Hallo Tom,

ich glaube der Fehler lag ganz klar bei mir. Kann es sein das ich überlesen habe das die Device Serial fix 10 Characters haben muss? Die hatte ich gekürzt.
Wünsche noch einen guten Rutsch ins Neue Jahr

Gruß
Werner

WernerSch, 2019-01-02:
Hallo Tom

genau das scheint es gewesen zu sein. Nur liegen anscheinend jetzt noch die alten Sensorleichen auf der CCU2 und ich finde sie nicht :-)

Gruß
Werner

TomMajor, 2019-01-03:
Hallo Werner,
mit der Serial length bin ich nicht sicher, würde da keine experimente machen und sie auf 10 char lassen, ich vermute einige HomeMatic Interna verlassen sich darauf.
Ist das Problem jetzt behoben oder was genau geht nicht?

WernerSch, 2019-01-03:
Hallo Tom

sieht so danach aus

Gruß
Werner