TomMajor/SmartHome #10: UNI-Sensor1: keine Temperatur vom DS18x20 wenn bmp/e280 vorhanden

harvey637, 2019-01-10:
Hallo Sir,
beim experimentieren mit dem UNI-Sensor ist mir aufgefallen, dass bei gleichzeitigen
Einsatz des DS18x20 UND des bmp/bme280 die Temperatur des bmp/bms280 gewinnt.

Das ist insofern unschön, da der ds18x20 ja nur ein Temperatursensor ist, wenn er schon 
vorhanden und gefunden ist sollte man auchseine Messwerte ausgeben. Eventuell ist der 
Kombisensor mit seiner Feuchtigkeits- und Druckmessung in einem offenen, aber abgedeckten Gehäuse,
der DS18x20 mit seinem (möglichen) wasserdichten Gehäuse an der eigentlichen Temperaturstelle.
Da der ds18x20 keine andere Aufgabe hat wird sein Ergebnis also ignoriert.

Das lässt sich leicht lösen, in dem die Zeilen 323-327 (DS18x20 Messung) hinter die 
Zeilen 348 (der SHT10 Messung und der bmp280 Messung) rückt.

Dann würde die interne Temperatur des BME280 in die Berechnung des Drucks eingehen (EnvironmentCalculations), der SHT wird sowieso nur genommen, wenn kein BME280 vorhanden 
und die einzige Aufgabe des DS18x20 wird als Temperatur des Ausgabepaketes verwendet.

ciao
Harvey

harvey637, 2019-01-11:
Als Erweiterung könnte man die Messung des DS18x20 nur dann verwenden, wenn tatsächlich
ein solcher gefunden wurde.
Dann wechselt die Quelle der Temperaturmessung automatisch, je nach dem, ob der 
DS18x20 gefunden wurde oder nicht, also ohne umdefinieren im Sketch.
Problemchen ist allerdings die Länge des Sketches im OTA Modus.

harvey637, 2019-01-12:
Hi Sir,

vollstes Verständnis, jeder kann das anpassen. Auch richtig, dass der Code eher droß ist. Allerdings habe
ich ihn mit Kürzen der Texte sogar MIT Debug und einem OTa-Bootloader unterbekommen, nut Text gekürzt.

Was aber keine "Codeänderung" ist, aber hilft wäre schon - aus meiner persönlichen Meinung - 
das Verschieben der ~5 Zeilen DS18x20 Messung hinter den BME-Teil.

Wenn man schon einen DS18x20 drin hat will man auch seine Messung sehen, sonst kann man ihn ja auch komplett draussen lassen. Wenn aber ein BME280 drin ist gewinnt immer der BME280.

Bei mir ist das so, dass der BME280 (und der Lichtsensor) IN einem wasserdichten Gehäuse (mit
durchsichtigen Deckel und Luftloch an der tiefsten Stelle für Druckausgleich und 
Feuchtiggkeitsmessung) eingebaut ist und der DS18x20 mit Edelstahlkopf an einer Rohrleitung
angeschraubt ist. Also habe ich genau nach deinen richtigen Kommentaren gehandelt und 
den DS18x20 Code nach hinten geschoben.

DIESEN Teil könntest Du sicherlich ohne Probleme umsetzen und damit den Code
nicht verlängern.

Ich kann prima mit meiner eigenen Änderungen leben (also der Codeverschiebung der 
DS18x20 Messung und der Reduktion der Debugausgaben), aber die Verschiebung erscheint 
mir auch für den Normalnachbauer ohne irgendwelche Nachteile und nur mit Vorteilen zu sein.

Aber es ist Dein Code, ich will nur mit meinen Erfahrungen und Stolperfallen helfen.
Daher schilesse ich dies als Issue, da für mich alles angepasst ist.

Danke aber trotzdem für das Lesen und immer weiter so!
Ciao
Harvey637

TomMajor, 2019-01-13:
@harvey637 
ok, wenn man den BME280 und den 18x20 beide drin hat kann es von Vorteil sein den 18x20 zu nehmen, wegen Einbausituation wie bei dir und weil die BME280 Temp. nicht die genauste ist.
Ich werde die Verschiebung ins Repo aufnehmen.

Ich möchte nur vermeiden das jede mögliche Sensorkombi eine issue wird, das soll jeder nach seinen Gegebenheiten anpassen :smile: 

Kannst du bitte mal ein Foto vom Gehäuse und besonders vom Luftloch verlinken?

harvey637, 2019-01-13:
Hi Sir,
das Gehäuse ist ein altes ELV Füllstandsensor, für Fotos ist es jetzt zu dunkel :-)
Das Gehäuse hat zwei Durchgänge, wo Kabelverprseeverschraubungen dran sind.
Durch eine Verschraubung geht die Leitung zum externen DS18x20, der wasserdicht
mit einer Edelstahlkappe eingeklebt ist.
Die andere Verschraubung ist einfach offen. Beides zeigt nacht unten.
In der Tat braucht die Feuchtigkeitsmessung (im Innernen mit BME280) etwas länger, bis sie 
halbwegs richtig anzeigt. Ich habe in der noch einen Original Ausssensensor Temp/Feuchtigkeit.
Aktuell halt so in der 90% ~ 95% Gegend. Irgendwann klebe ich noch einen Teebeutel(Gaze) von innen
hinter das Loch, obwohl die Insekten bestimmt protestieren!

Noch ein Grund, den DS18x20 nach "hinten" zu schieben: Auch wenn kein BME280 drin ist (Dummywerte!) überschreibt er damit die echten gemessenen Werte des DS18x20. Genau 
genommen weiß ich garnicht, wann man sie überhaupt bekäme :-)

By the Way und eigentlich nicht in diesen Isse:
Ich teste auch die Battery-Variante mit eienr Batterie und einem StepUP, also Batteriemessung
über A0 und D9 mit 150K/100K Spannungsteiler. Batteriespannung wird auch korrekt ermittelt.
ALLERDINGS lässt das Addon als LowBat nur minimal 1.8V zu. Ich denke, bei EINER Batterie
sollte low bei 1.0 und crit bei 0.8 liegen, da dann der StepUp zusammenbricht.
Lässt sich aber nicht einstellen, da die CCU sofort die Konfiguration des Sketches überschreibt.
Daher wird der Sensor sofort als LowBat gemeldet, obwohl ein geladener Akku mit 1.23V drinsteckt.
Ich kann das selbst im Addon für mich korrigieren. Mir fällt auch nicht ein, wie das Addon auf die Varianten der Spannungsversorgung (StepUp, StepDown, 2*1.5V....) reagieren könnte.
Also muss man die richtigen Schwellwerte manuell eintragen (und es nicht vergessen). dabei wird man
aber bei BatLow MINIMUM 1.8V blockiert.
Ansonsten klappt wunderbar - Danke!
ciao
Harvey627

harvey637, 2019-01-15:
Hmmmm, ich hatte den Eindruck, trotz "//#define bme280" nicht die Temperatur des angeschlossenen
und in der Debugausgabe via FTDI richtigen sichtbaren DS18x20 Temperatur doch nur eine 
dummy-Temperatur in die CCU-Anzeige zu bekommen.
Andererseits bin ich immer selber schuld :-) Und wenn der DS18x20 "hinten" steht ist immer gut!

Low-Bat ... super, danke! Habe ich manuell schon geändert und nach Neuanlernen klappt es.
Eventuell sollte der Bereich "großzügig" sein, da es ja auch ein 12V/5Ah Akku (habe ich 12 Stück
aus USV von + StepDown) sein kann. Natürlich muss der Spannungsteiler dann anders sein.
Bei einen Universalgerät (was ich gerne zu Testen verwende) sind ja diverse Spannungsversorgungen
möglich, auch mit der PLHT-Platine.

Luftloch: bei mir einfachst und kostenlos. Ich hatte immer Probleme mit Levelsensor, zweimal
defekt wegen Kondenswasser (jetzt Asksinpp Nachbau :-) Irgendwie vielleicht die Platine
wasserfest versiegeln (es gibt Spray), aber der Sensor BME280 (und Taster und Batteriehalter)
muss ja frei bleiben. Im Aussenbereich gibt es ja immer Kondensation im Gehäuse.
Auch mit Silicagel gespielt, aber wenn man Feuchte messen will.... Vielleicht muss man NUr
den Sensor an die Oberfläche bringen und den Rest komplett dicht versiegeln.... Langzeit?
erstmal Danke und ciao
harvey637

TomMajor, 2019-01-16:
Klingt plausibel mit der Feuchtigkeit bzw. das der Nylon übertrieben ist.
Wir lassen die issue mal offen bis ich die Änderungen committed habe und ein Foto der Gaze/Teebeutel Lösung würde mich immer noch interessieren :smiley:

TomMajor, 2019-02-03:
Habe harvey637's Idee mit dem Nylonstrümpfen hier zitiert:
[FHEM Forum](https://forum.fhem.de/index.php/topic,20620.msg899233.html#msg899233)
und hier gibt es Fotos von einer praktischen Ausführung:
[FHEM Forum](https://forum.fhem.de/index.php/topic,20620.msg900396.html#msg900396)