TomMajor/SmartHome #11: Stromsparen mit den Sensoren

harvey637, 2019-01-14:
Hi Sir,
(ich bin ganz lieb und schreibe nur auf, was mir auffällt:-)

Einige Sensoren (z.B. BME280) gibt es auch in "echten" 3V3-Versionen ohne LDO, da braucht wohl nix
mehr optimiert werden.
Bei dem Helligkeitsmesser (MAX44009) ist mir ein Lötklecks auf der Oberseite aufgefallen.
Möglicherweise dient der zum Umsetzen der Spannungquelle. Vielleicht schöner als
der Lötklecks über dem Widerstandsarray .... Ich schaue mir das mal an.
cu
Harvey637

TomMajor, 2019-01-14:
Die Lötbrücke beim MAX44009 ist vermutlich die I2C Addr. Auswahl.

harvey637, 2019-01-15:
ok, danke. 
Sieht in der Tat so aus, als würde der mittlere Pin an A0/Pin3 des max44009 gehen. 
A0 = Address Select. Pull high to select address 1001 011x or low to select address 1001 010x
Manual PDF Seite 16, Tabelle 10.

Also LDO ablöten und Widerstandsnetzwerk verbinden wie beschrieben. Eventuell macht es Sinn, die Addressänderung im der AskSinpp Seite zu erwähnen?

ciao
Harvey637

closed 2019-01-15 by harvey637