TomMajor/SmartHome #14: Helligkeit mit hoeherer Aufloesung

HMSteve, 2019-03-09:
Hallo Tom,

waere es moeglich, die Uebergabe der Helligkeit in mLux vorzunehmen, so dass auf der CCU (bis zu) drei Nachkommastellen bei der Lux-Angabe zur Verfuegung stehen? So koennte man die super Empfindlichkeit des MAX44009 auch in der Daemmerung voll ausnutzen.
Die vorgesehene Variable genuegt als 32bit int dafuer ja schon, so dass die Anpassungen marginal sind. Habe das durch Anpassung des AVR-Codes und AddOns fuer mich hingebastelt, wuerde aber lieber beim "offiziellen" AddOn bleiben.

Danke,
Stephan

TomMajor, 2019-03-10:
Hallo Stephan,

prinzipiell ja, ich habe nur momentan ein Zeitproblem mit solchen requests. 

Beim TSL2561 hat man sicher nichts von den Nachkommastellen und beim MAX, ok vielleicht wäre 1 sinnvoll bei Dunkelheit, aber 3? Ist das nicht alles nur noch Rauschen? Machst du wirklich SmartHome Aktionen auf Basis der 3. Kommastelle?

Ich werde den request im Hinterkopf behalten und die issue offen lassen, kann aber momentan nicht sagen wann und ob ich das mache (und ob es wirklich 3 Kommastellen dann werden, ohne Gegenbeweis finde ich das eher zweifelhaft :smile_cat:

HMSteve, 2019-03-11:
Hallo Tom,

 

ein bisschen Freude an dem, was mit dem MAX44009 geht, ist schon dabei: Warum 4-5 der 22bit DR verschenken?

 

Aber vielleicht gehen echte Beobachtungsergebnisse als Beweis durch: Anbei die Aufzeichnung der letzten Nacht mit sampling etwa alle 3min. Man sieht schoen, dass die Beschraenkung auf Ganzzahlen ziemlich genau die gesamte Daemmerung von 0 bis -6deg Sonnenstand abschneidet (ca 30min zwischen 1Lux und 0). Genau waehrend dieser Zeit moechte ich das Einschalten bestimmter Leuchten triggern – weder schon bei Tage noch erst im voellig Dunklen. 

 

Man sieht auch das sehr geringe Rauschen: Die beiden peaks am Abend sind meine Aussenbeleuchtung, in deren Naehe der Sensor noch liegt, ansonsten kippt nur das LSB (45mLux) gelegentlich, und selbst das ist a.G. der Fassadenbeleuchtung meiner Nachbarn womoeglich noch plausibel- richtig dunkel ist es hier nicht. Insofern enthaelt selbst das LSB, erst sichtbar in der 2. bis 3. Nachkommastelle, vermutlich nicht nur reinen Unsinn. Verhalten passt also zum Datenblatt.

 

Drei Stellen aber auch deshalb, weil Runden jeder selber kann, jedoch Erweitern nicht, und die 32 Bit zur Uebertragung an die CCU bereits ungenutzt „herum liegen“. Es waere nach meinem sehr oberflaechlichen Verstaendnis also nur je ein Skalierungsfaktor im xml und AVR-Code anzupassen, was vielleicht mit geringem Aufwand in der naechsten (oder einer spaeteren) Version nebenbei geschehen koennte?

 

Viele Gruesse,

Stephan

 

 

PS: Als naechstes liebaeugele ich uebrigens mit dem MLX90614 als Bewoelkungssensor mittels Temperaturdifferenz sowie dem VEML6075 (wohl ein grosses Projekt, fuer mein Kenntnis-/Zeitbudget eher unrealistisch, aber wohl deutlich moderner als der VEML6070).

TomMajor, 2019-03-12:
Echte Beobachtungsergebnisse würden def. als Beweis durchgehen :smile: 
Deine Textbeschreibungen der Messergebnisse klingen verführerisch, ich sehe nur leider keine Diagramme

HMSteve, 2019-03-12:
Oh, da hing ein jpeg einer Nacht an der Mail, das hat Github offenbar heraus gefiltert. Es liegt nun hier, zusammen mit einer weiteren Nacht als Indiz fuer die Reproduzierbarkeit und einem Vergleich zum HM-Wettersensor: https://www.dropbox.com/sh/dci6f1c7c1osody/AABiYkyr-eTPCrjZeHiSH6e1a?dl=0

 

Viele Gruesse,

Stephan

TomMajor, 2019-03-13:
Danke für die Diagramme.
Was ich nicht ganz verstehe, du zeigst in MAX44009_Night2.JPG im interessanten Bereich der Dämmerung ca. 9 verschiedene Lux Stufen zwischen 0-2 Lux.
Ok, du hast 3min sampling intervall, aber was passiert wenn du dir mal die Werte mit z.B. 1sec sampling plottest?
Werden das dann signifikant mehr Stufen? Sonst verstehe ich nicht wieso du für 9 Stufen 3 Kommastellen brauchst. Vielleicht habe ich aber auch die Messung noch nicht voll begriffen :grey_question:

HMSteve, 2019-03-18:
Hallo Tom,

 

wie gesagt sind 3 Dezimalstellen natuerlich streng genommen nicht noetig, um die 21 Stufen bis 1 Lux aufzuloesen, aber fuer null Zusatzaufwand und –kosten einfach „nett“, da man dann die 45mLux-Stufen direkt ablesen kann, die der MAX lt Datenblatt liefern sollte und auch mit Bravour liefert. Habe dazu mal alle 10s gesampled, siehe anbei (hat etwas gedauert, da ich erst ein neues xml mit einem kleineren Minimum fuers Update-Intervall einspielen musste). 

Ueber die absolute Genauigkeit der Messung rede ich hier nicht, aber ueber das extrem saubere, i.e. rauschfreie Verhalten des Sensors. Da bin ich echt begeistert, wenn ich das mit meinen Versuchen mit irgendwelchen analog ausgebenden Sensoren vor ca 10 Jahren vergleiche.

 

Vielleicht habe ich aber auch Dein Argument gegen die 3 Dezimalstellen noch gar nicht vestanden, mein Gedanke war, von 32 sowieso uebertragenen bits nicht 10, sondern nur 7 ungenutzt zu lassen, was m.M.n. keinen Schaden anrichtet?

 

Viele Gruesse,

Stephan

TomMajor, 2019-03-20:
Ich sehe wieder keinen Anhang bzw. die neuen Messwerte.
Ich möchte möglichst wenig Änderungen am AddOn haben da dies immer etwas Aufwand bedeutet, build, test usw. + die Änderungen sollten sinnvoll sein. Bei 45mLux Stufen wären max. 2 Kommastellen für mich sinnvoll, außerdem zeigt das Gerät im WebUI nur 2 Kommastellen an.
Aber das kann und soll auch jeder für seine Bedürfnisse anpassen, ich kann aber nicht jede spezielle Anpassung universell im AddOn vorhalten.

HMSteve, 2019-03-20:
Ist verstaendlich. Falls Du irgendwann doch 2 Dezimalstellen einbaust, werde ich gern auf das „offizielle“ addon zurueck schwenken.

Die feiner aufgeloeste Grafik liegt nun uebrigens auch unter dem alten Dropbox-Link.

TomMajor, 2019-03-20:
Du meinst Morning18Mar.JPG, oder. Dort sieht man ca. 21 Stufen zwischen 0..1 Lux was ziemlich gut der Auflösung im Datenblatt entspricht. 
Schon erstaunlich wie gut der MAX im Dunkeln bzw. bei Dämmerung ist, danke für den Hinweis.
2 Stellen werde ich einbauen, wahrscheinlich noch im März. Ich lasse die issue solange mal offen.

HMSteve, 2019-03-21:
Prima, vielen Dank, und ja, diese Datei meinte ich.

TomMajor, 2019-04-09:
Im Zuge der gerade fertig gewordenen Umbauarbeiten für 'Benutzerspezifischen Sensordaten' und VEML6070 support war eine neue Firmware- und AddOn-Version nötig, bei der Gelegenheit habe ich auch gleich die 2 Nachkommastellen für den MAX44009 gemacht.

closed 2019-04-09 by TomMajor