TomMajor/SmartHome #15: HB-UNI-Sensor - add TEMP6000 support [solved as patch]

harvey637, 2019-03-10:


TomMajor, 2019-03-10:
Was ist der Vorteil des TEMT6000 gegenüber den beiden anderen unterstützten Helligkeitssensoren?
Die Helligkeit wird aktuell im AddOn als LUX übertragen, % ist schlecht (und was wäre 100% ?)

harvey637, 2019-03-11:
Hi, der Preis :-) Und nicht jeder Sensor braucht eine Auflösung auf drei Nachkommastellen, etwa
Bewegungssensor. Darüber hinaus ist die Behandlung simpel.
Da mir eine nicht lux-genaue Info reichte und der Einbau in den Sketch minimal ist habe
ich es einfach mal gemacht. Und der UNI-SENSOR ist halt einfach erweiterbar.
Ausserdem einfach ein weiterer einbindbarer Sensor.
Part2: die geänderten Dummywerte zeigen besser, wenn Sensor wirklich nicht da ist.
Part3: DDEVINFO(sdev) ... ist halt schon vorhanden
Wie gesagt, alles nur Vorschlag, kann jederzeit gerne weg, und es steht ja drin "... kann selbst angepasst werden ..."
PS: Infos zu Fuses und Bootloader finde ich gut, ich hatte auch mal für mich die Verwendung des OTA-Bootloader vereinfacht, das ist auch noch etwas intransparent, aber lernbar.
Danke und ciao
Harvey

TomMajor, 2019-03-17:
ok, alles klar. Wenn jemand den Sensor braucht kann er das ja entsprechend modifizieren.
Ich nehme die Änderung mal nicht mit aktiv in den Sketch auf weil der physikalische Wert nicht zum AddOn xml passt.
Danke für den Beitrag.

closed 2019-03-17 by TomMajor