TomMajor/SmartHome #16: DS18X20

chkbln, 2019-03-19:
Hallo Tom
kann man an dem Uni-Sensor1 mehr als einen DS18B20 anschliessen?
Viele Grüße

TomMajor, 2019-03-20:
Prinzipiell ja, 1-Wire ist ein Bus wo mehrere Sensoren existieren können.
Schau dir mal dieses 8-fach DS18B20 Projekt an
https://github.com/jp112sdl/HB-UNI-Sen-TEMP-DS18B20
Im Uni-Sensor1 kann ich das nicht universell für alle machen da ja das AddOn für 8-fach sehr speziell wird.

chkbln, 2019-03-20:
Hallo Tom,
vielen Dank für deine Antwort.
Den HB-UNI-Sen-TEMP-DS18B20 habe ich schon getestet, aber leider fehlt die Integration in FHEM.
Bei dir ist ja die Integration in FHEM vorhanden. Deshalb die Anfrage.
Im Forum von FHEM sind bezügl. der Intergation vom HB-UNI-Sen-TEMP-DS18B20 schon Anfragen gewesen. Aber es gibt keine Lösung.
Gruß

TomMajor, 2019-03-24:
Verstehe dein Problem, kann dir da aber nicht helfen.
Den UniSensor1 kann ich wegen dem AddOn nicht mal eben auf 8 Temp.kanäle umstellen.
Lösung für dich wäre: Beschäftigung mit Perl/FHEM um die vorhandenen Sensor Perl Skripte an Jeromes HB-UNI-Sen-TEMP-DS18B20 anzupassen.

chkbln, 2019-03-25:
Die Lösung mich mit Perl zu beschäftigen und selbt eine integration in FHEM zu programmieren habe ich für mich ausgeschlossen. Die Zeit wird im Alter kostbar. Deine Zeit sichlich genauso. Ich wollte halt wissen ob das vorhanden genutzt werden kann.
Gruß

closed 2019-03-25 by chkbln

TomMajor, 2019-03-25:
Anderes Gerät mit anderer payload erfordern Anpassungen im Perl Skript, da führt leider kein Weg dran vorbei. :crying_cat_face: