TomMajor/SmartHome #19: Verwendung customData (UV-Index) in IOBROKER

harvey637, 2019-04-23:
Hallo Sir Tom Major,
ich habe in dem aktuellen Addon custom_xml aktiviert und kann dann mit neu angelerntem Sensor mit der Firmwareversion auch den Datenblock customData sehen, sowohl im arduino-Monitor als auch in standard-HM-Webgui und im HW-Webgui.
Also da steht in der Gerätesicht ein Feld "[UV-Index]" (die [ ] habe wohl mit der Übersetzung zu tun???) und darin der richtige Wert, z.B. 3.
Egal wie ich in IOBROKER das rpc-Objekt lösche und die rpc-Instanz neu synchronisiere, 
dieser Wert (customData / UV-Index) taucht nie auf. Entsprechend kann ich auch keine Daten in die History aufzeichnen.
Ne Idee? Oder ist der Sync mit iobroker firmwareversions-unabhängig, also immer die "base" firmware, die dann das Element customData/UV-Index garnicht hat?
Liegt hier eventuell ein Denkfehler bei mir vor?
cu
Harvey

TomMajor, 2019-04-23:
die [ ] bedeuten dass es kein nativer HM Messgrößentyp ist und es keine Übersetzung gibt.
Stört mich normalerweise nicht, übersetzten könnte man das in der richtigen Language Datei, der ioBroker nimmt das aber nicht her, deswegen mache ich keine Übersetzungen mehr und lebe mit den [].

rpc neu syncen dass die geänderte Konf. erscheint, dazu habe ich gerade was im orangen Forum geschrieben.

harvey637, 2019-04-23:
Hi,
wie im Forum geschrieben .... klappt nicht.
Fehlende Übersetzung, kann ich mit leben. Manchmal finde ich die mehrfachen Antworten (Channels?) 
wie z.B. bei dem "ds18x20 8fach" ganz gut, da ist der Messwert immer "Temperatur", aber der Datenpunkt 
hat einen eigenen (änderbaren) Namen, so kann ich aus "[Temperatur_1]" (Messgrößentyp) einen
Datenpunkt "SENSOR005.Solar_vom_Dach" (mit der Messgröße Temperatur) machen.
Vielen Dank, dass ihr mir dauernd helft!
cu
Harvey

closed 2019-04-23 by harvey637