TomMajor/SmartHome #23: Erweiterung um Temperatur und Feuchtesensor SHT21 / Si7021

chianti2000, 2019-06-23:
Ist es möglich, das du den den Temperatur- und Feuchtesensor SHT21 / Si7021 einbindest.
Soll geringere Fehlertoleranz insbesondere bei Feuchte aufweisen, wo der BME 280 ja etwas problematisch erscheint. Kann dann mit dem BMP 280 im Unisensor arbeiten.

TomMajor, 2019-06-24:
Prinzipiell ja, hätte sogar einen unbenutzten SHT21 rumliegen.
Momentan nur ein Zeitproblem, weiß nicht wie dringend es wäre.

chianti2000, 2019-06-25:
Kein Problem hat noch Zeit, wäre aber gut, wenn es nicht in Vergessenheit gerät. Laut Jerome  gibt es in der AskSinPP den Sensor schon. 
https://github.com/pa-pa/AskSinPP/blob/master/sensors/Si7021.h
https://github.com/jp112sdl/Beispiel_AskSinPP/tree/master/examples/HM-WDS40-TH-I-SI7021
Würde ihn aber gerne im UNI-Sensor und nicht Stand -Alone nutzen wollen.

TomMajor, 2019-06-26:
ich schaue ob ich den SHT21 in den nächsten Wochen eingebaut bekomme und lasse den thread mal so lange offen.

chianti2000, 2019-06-29:
Danke

TomMajor, 2019-08-09:
Support für SHT21 / Si7021 hinzugefügt.

closed 2019-08-09 by TomMajor

chianti2000, 2019-08-09:
DANKE! 👍