TomMajor/SmartHome #24: HB-UNI-Sensor1 - Sens_BH1750

wolwin, 2019-06-30:
Hi, da der BH1750 Helligkeitssensor nicht implementiert ist und ich gerade DIESEN Sensor gekauft hatte, habe ich die (momentan aktuelle) HB-UNI-Sensor1.ino angepasst und eine neue Sens_BH1750.h erstellt. Nach dem Studium der relevanten (aber nicht ganz fehlerfreien) BH1750 Libs habe ich mich für die AS_BH1750 Lib entschieden:
https://github.com/hexenmeister/AS_BH1750

Interesse ?? Wenn ja, bitte melden und ich schicke Dir die beiden Files …

TomMajor, 2019-07-01:
Hi Wolfram,
gerade neulich hatte jemand im HM Forum nach diesem Sensor gefragt
https://homematic-forum.de/forum/viewtopic.php?f=76&t=43887&start=60#p513953
du bist nicht zufällig dieser user?
Anyway, schicke mir gern die Files, ich baue die bei Gelegenheit ein, kann aber momentan zeitlich nichts versprechen.
Danke,

wolwin, 2019-07-02:
Hi Tom, nö - ist ein anderer User … habe alle Stromspareffekte von Dir in meinem Testaufbau umgesetzt - auch auf dem BH1750 kann man den LDO auslöten und eine Verbindung Vin-Vout herstellen - ich erreiche dann die gewünschten 5uA Ruhestrom. Lediglich wenn ich den DS18B20 anschließe, liegt der Ruhestrom bei 42 uA (ist das so richtig?) !! Probiere deshalb z.Z. eine An/Abschaltung der Sensoren über Port D4 mit einem N-MOSFET aus … funktioniert prinzipiell so weit … Intialisierung der Sensoren muss noch angepaßt werden.

Hier jedoch erst einmal die beiden Files … ist eigentlich nichts weiter anzupassen .. wirf einfach mal einen Blick drauf - Danke!

[HB-UNI-Sensor1_20190630.zip](https://github.com/TomMajor/SmartHome/files/3349890/HB-UNI-Sensor1_20190630.zip)

[Sens_BH1750.zip](https://github.com/TomMajor/SmartHome/files/3349885/Sens_BH1750.zip)

TomMajor, 2019-07-02:
Hi Wolfram,
danke für die Files, schaue ich mir demnächst an und gebe noch mal Rückmeldung wenn das online ist.

42uA für DS18, nein, nicht im Sleep, da stimmt was nicht.
Aus meiner Erinnerung braucht der weniger als 1uA
Beim Bild hier mit 3,75uA ist der DS18 mit drin:
https://github.com/TomMajor/SmartHome/tree/master/Info/Ruhestrom#%C3%BCberpr%C3%BCfung-des-avr-ruhestroms-power-down-mode

TomMajor, 2019-07-13:
Support für BH1750 ist integriert.
Danke an @wolwin

closed 2019-07-13 by TomMajor