TomMajor/SmartHome #25: AS3935 Sensor

Michel83, 2019-10-06:
Hi, ist es möglich ein AS3935 Blitzsensor mit aufzunehmen? 

Gruss und super Arbeit

TomMajor, 2019-10-06:
Habe mich noch nicht mit diesem beschäftigt, prinzipiell wahrscheinlich schon. Stromverbrauch ist ev. ein Thema da der Sensor wohl immer an sein muss?

TomMajor, 2019-10-09:
Stromverbrauch ist ca. 70uA, das müsste für mehrere Jahre reichen. Man könnte also wahrscheinlich einen Blitzdetektor mit dem Unisensor bauen.
Falls mir jemand einen Sensor sponsert schaue ich mir das im Winter mal an..

Michel83, 2019-10-11:
Hi, danke für die Antwort.

Ich würde ein Sponsoren :-)

TomMajor, 2019-10-11:
ok, wenn es ok für dich ist dass es ein paar Wochen dauern kann bis ich damit anfange (irgendwann im Nov.) können wir das gerne machen.
Weiteres am besten über PN im HM oder FHEM Forum.

Michel83, 2019-10-12:
PM im HM Forum geschrieben

Grüße

TomMajor, 2020-02-16:
Sensor bekommen, vielen Dank, Thema ist in Arbeit.

closed 2020-02-16 by TomMajor