TomMajor/SmartHome #30: HB-UNI-Sensor1 - Sens_SHT31

wolwin, 2019-12-04:
Hi Tom,
nachdem ich nur Ärger mit dem BME280 im Aussenbereich habe/hatte, wollte ich jetzt mal den SHT31 (mit PTFE-Membran von Adafruit) ausprobieren. Dazu habe ich Deine aktuelle Version 1.16 um die SHT31 Aufrufe erweitert - die geänderten / neuen Files sind im RAR-Archiv beigefügt (und live so getestet). Vielleicht kannst Du den SHT31 ja übernehmen ...
Grüße
Wolfram

[HB-UNI-Sensor1_SHT31.rar.zip](https://github.com/TomMajor/SmartHome/files/3921671/HB-UNI-Sensor1_SHT31.rar.zip)

TomMajor, 2019-12-04:
Hi Wolfram,
Danke. Ich schaue es mir die nächsten Tage an und kann das dann mit übernehmen.
Das ist dann der hier vermute ich:
https://www.adafruit.com/product/2857
Du hattest Probleme mit Tau bzw. Kondenswasser beim BME?

wolwin, 2019-12-05:
Hi Tom,
genau der Sensor ... wichtig ist das F wg. der PTFE Folie - bin gespannt, ob der SHT eine bessere Langzeitstabilität im Außeneinsatz hat, als der BME280. Für den BME280 habe ich schon ein eigenes Wettershield mit eingebautem Lüfter konstruiert und gebaut (siehe: https://github.com/wolwin/WW-myHOME/tree/master/myHOME%20-%20Hardware/BME-Shield) - trotz PTFE Abdeckung ist vor zwei Wochen von zwei Shields die Weatherman Einheit abgesoffen - die zeitgleich installierte Feinstaub Einheit läuft. Insgesamt habe ich im letzten Jahr insgesamt drei Ausfälle von fünf BME280 Sensoren im Garten zu beklagen ... aber das jetzt ein bischen OT ... ich habe inzwischen Dein  UniSensor Design mit eigener Platine (Besückung Standardbausteine, 'alle' Anschlußmöglichkeiten, MOSFET geschaltetes GND, Stepdown für Batteriebetrieb) gepimpt - zusätzlich 3D-Drucker Multi-Design für den Einbau in OBO-Boxen incl. Sensoren-Wetterhäuschen ... vielleicht sollte ich mal einen Stand ins Forum setzen.
PS: ... und alles nur wg. zwei alten Gardena Sensoren, die ich an den UniSensor bringen wollte ...

TomMajor, 2019-12-06:
Hi Wolfram,
sehr interessant, dein Wettershield.

Ich habe den SHT31 support commited.

Ein Fehler war noch drin, der ctor von Adafruit_SHT31 erwartet entweder nichts oder ein alternatives Wire-Interface
https://github.com/adafruit/Adafruit_SHT31/blob/master/Adafruit_SHT31.h#L56
du hattest dort I2C_ADDR.
`: _sht31(I2C_ADDR)`
->
`b:\Temp\arduino_build_929356\sketch\Sensors/Sens_SHT31.h:33:22: warning: invalid conversion from 'unsigned char' to 'TwoWire*' [-fpermissive]`

Und diese Zeile war imho nicht nötig, habe sie rausgenommen
`_sht31 = Adafruit_SHT31();`

Ist jetzt ungetestet da ich den SHT31 nicht da habe.

wolwin, 2019-12-06:
Hi Tom,
Danke für Deine Korrekturen - habe sie gerade getestet: funktioniert !!
Wenn Du bitte noch die von mir vergessene Kommentarzeile 34 in 'Sens_SHT31.h' löscht, dann ist alles perfekt ...

DANKE !!

wolwin, 2019-12-06:
... meinte natürlich Zeile 44 ...

TomMajor, 2019-12-06:
ok Zeile ist entfernt.
geht morgen online.
Danke für deinen SHT31 Beitrag :smiley_cat:

closed 2019-12-06 by TomMajor