TomMajor/SmartHome #32: HB-UNI-Sensor1 - Sens_BMP280_MOSFET

wolwin, 2019-12-30:
Hi Tom,
z.Z. stelle ich gerade meine PCBs und INOs für Github zusammen - dabei biin ich dann auch über meine Sensor-Tests 'gestolpert'. 
Für den BMP280 habe ich Dein momentanes INO-Script angepasst und entsprechend erweitert - das BME280 Modul verarbeitet zwar auch den BMP280, aber es kamen andere Werte heraus, als wenn ich die spezielle Adafruit Library nehme - diese erschienen mir im Vergleich mit dem SHT31 plausibler die Werte aus dem BME280 Modul.
Ausserdem habe ich eine kleine N-MOSFET Ansteuerung als Sensor Klasse eingefügt. Damit ist es möglich die Sensoren während des Sleep-Modus 'abzuschalten' - funktioniert bei mir wunderbar (spart auch das Auslöten der LDOs ...).
Vielleicht kannst Du die Änderungen ja übernehmen ... siehe RAR-Archiv mit den Änderungen  :-)
Gruss
Wolfram

[HB-UNI-Sensor1_BMP280_MOSFET.zip](https://github.com/TomMajor/SmartHome/files/4010412/HB-UNI-Sensor1_BMP280_MOSFET.zip)

TomMajor, 2020-01-01:
Hey Wolfram,

Danke.

bin nicht sicher wie du das mit der BME Lib meinst, willst du sagen die beim UniSensor verwendete
https://github.com/finitespace/BME280
kann mit der BMP Variante nicht richtig umgehen?
Würde mich wundern, ich dachte die beiden Typen sind bis auf die relative Luftfeuchte gleich?

Die Mosfet Abschaltung für den Sensor ist für mich nicht so interessant da ich die LDO auslöte wenns drauf an kommt. Behalte das Projekt lieber in deinem github.

wolwin, 2020-01-02:
Hi Tom,

> kann mit der BMP Variante nicht richtig umgehen?
> Würde mich wundern, ich dachte die beiden Typen sind bis auf die relative Luftfeuchte gleich?
... dachte ich auch ... war jetzt eher eine quantitative Einschätzung, weil der BMP andere Werte brachte als der SHT31 - habe für den BMP die Adafruit-Library benutzt, da Adafruit zwischen BME und BMP unterscheidet ... mit der BME Adafruit Library entsprachen die Werte den SHT31 Werten. Ich habe das dann nicht weiter untersucht .... natürlich kann man auch die BME Library nutzen (die BME Debug Informationen ins Sens_BME280.h habe ich für eine Unterscheidung zwischen BME und BMP ergänzt).   

> Die Mosfet Abschaltung für den Sensor ist für mich nicht so interessant da ich die LDO auslöte wenns drauf an kommt. Behalte das Projekt lieber in deinem github.

OK - habe eine (konventionelle) 'HB-UNI-Sens-X' und eine 'HB-UNI-Mini-X' Platine entwickelt, auf der u.a. auch die MOSFET Abschaltung optional angelegt werden kann (werde die PCBs vsl. nächste Woche auf Github als OpenSource veröffentlichen). Mein Ziel war - zugegeben - nicht ganz uneigenützig - eigentlich wollte ich mir den eigenen Github Repo Aufwand für die 'paar Zeilen' sparen ... Dein INO-Script (aus dem RAR-Archiv) läuft auf den PCBs sofort ohne jegliche Änderung. Vielleicht kannst Du es Dir ja noch einmal überlegen, die Zeilen 'undokumentiert' zu übernehmen. Ich würde bei den PCBs auf die Mosfet Konfigurationsmöglicht hinweisen. DANKE (für Deine Geduld mit mir ...) !!!

TomMajor, 2020-01-03:
ok, ich schaue mir das mit dem Mosfet noch mal an, kann aber etwas dauern, momentan wenig Zeit.

wolwin, 2020-01-05:
Hi Tom,
das mit den unterschiedlichen Werten vom BMP280 BME280 hat mir keine Ruhe gelassen und ich habe heute mal eine längere Debug-Sitzung abgehalten und alle Variationen durchprobiert ... mit einem eindeutigen Ergebnis: sobald man den BMx die Spannung weggenommen hat (Mosfet), muss er neu initialisiert werden - andernfalls kommen falsche Werte. Eigentlich logisch .... mein Fehler ... 
Die Lösung ist ebenfalls ganz einfach .... in der Sens_BME280.h muss nur eine init-Zeile in der measure- Funktion eingefügt werden. Die von mir vorgeschlagene Änderungen der BMP280 Erweiterung kann damit ersatzlos entfallen - es müssen nur ein paar Zeilen in der Sens_BME280.h ergänzt werden.
Ich stelle Dir das dann gleich hier noch ein ... hoffentlich bist Du noch nicht aktiv geworden ...

wolwin, 2020-01-05:
So ... hier die Änderungen ...
[HB-UNI-Sensor1_BME280_MOSFET_Add.zip](https://github.com/TomMajor/SmartHome/files/4023548/HB-UNI-Sensor1_BME280_MOSFET_Add.zip)

TomMajor, 2020-02-16:
Sorry für das Delay. Ich möchte das Thema mit der Mosfet Abschaltung hier nicht weiter verfolgen.
Würde vorschlagen die Anpassungen in deinem github fortzuführen. Danke.

jp112sdl, 2020-02-16:
Wie viel spart man eigentlich durch die elektronische Abschaltung des Sensors?

https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme280-ds002.pdf
> 0.1 μA in sleep mode

Lohnt sich da noch der Aufwand mit dem MOSFET?

TomMajor, 2020-02-16:
ich glaube wolwin geht es um den LDO bei den Breakout Boards, 5..6uA
https://github.com/TomMajor/SmartHome/tree/master/Info/Ruhestrom#ruhestrom-mit-sensor-boards

ich löte sie aber lieber aus wenn es drauf an kommt..

wolwin, 2020-02-19:
Genau (Breakoutboards, LDOs ... ) - ist definitiv kein Muss (aber ein Kann) - man benötigt ja auch die Hardwareunterstützung dazu ... mich haben in Teilen echt die Unterschiede zw. Doku und Realität genervt ( ... Thema Fake), sodass ich inzwischen im Sleepbereich gerne alles mit dem MOSFET abschalte ...

TomMajor, 2020-02-20:
Was man auch noch beachten sollte:
Wenn man die Sensor Breakout Boards über Mosfet abschaltet sollten die I2C pins auf Low liegen sonst kommt es ggf. zu Stromabfluss bzw. Verletzung der Spec (I/O Spannung ohne Vcc).
Da diese Pins aber normalerweise unter Kontrolle der Arduino Wire Lib liegen erfordert das ggf. einen unschönen Hack.

closed 2020-02-22 by TomMajor