TomMajor/SmartHome #33: AskSin-Analyzer-XS Gerber 404

Yannik, 2020-04-19:
Hi,
ich wuerde mir gerne einige der von dir designten AskSin-Analyzer-XS pcbs aetzen lassen.
Leider gibt der [link](https://github.com/TomMajor/SmartHome/blob/master/PCB/AskSin-Analyzer-XS/Gerber) zu den gerber files eine 404.

TomMajor, 2020-04-19:
Hi Yannik,

das ist Momentan Absicht.

> Christoph und ich haben bzw. werden davon eine Prototypenbestellung machen.
Ich würde empfehlen auf den ersten Aufbau eines Gerätes zu warten ob alles funktioniert bevor das ggf. jemand nachbestellt, z.B. die Sache mit dem 4-Kanal Levelshifter-Board ist für mich Neuland, das setze ich zum ersten Mal ein.

https://homematic-forum.de/forum/viewtopic.php?f=76&t=56395&start=170#p573219

ich hoffe es dauert nicht mehr lange bis der Funktionsnachweis da ist, wir warten momentan noch auf die Platine.

TomMajor, 2020-05-10:
Test der Schaltung/Platine war erfolgreich, die Gerberdateien sind jetzt verfügbar.

closed 2020-05-10 by TomMajor