TomMajor/SmartHome #34: Fehler bei der Kompilation HB-Uni-Sensor1

HMelzer, 2020-04-22:
Bei der Kompilation des Programmfiles tritt folgender Fehler auf:

HB-UNI-Sensor1\HB-UNI-Sensor1.ino: In member function 'virtual void WeatherChannel::trigger(as::AlarmClock&)':

HB-UNI-Sensor1:324:22: error: 'as::Channel<Hal, as::List1, as::EmptyList, as::List4, 6, SensorList0>::DeviceType {aka class as::Device<Hal, SensorList0>}' has no member named 'broadcastEvent'; did you mean 'broadcastPeerEvent'?

             device().broadcastEvent(msg, *this);

                      ^~~~~~~~~~~~~~

                      broadcastPeerEvent

exit status 1
'as::Channel<Hal, as::List1, as::EmptyList, as::List4, 6, SensorList0>::DeviceType {aka class as::Device<Hal, SensorList0>}' has no member named 'broadcastEvent'; did you mean 'broadcastPeerEvent'?

TomMajor, 2020-04-22:
bitte auf die neueste AskSinPP aus dem master Branch aktualisieren.

HMelzer, 2020-04-23:
Hallo,
danke für die schnelle Antwort. Und das war auch die Lösung.

Ich habe jedoch noch ein anders Problem festgestellt:
Wenn ich das _Update HB-UNI-Sensor1 AddOn to version 2.04_ als Zusatzsoftware in der CCU installiere, bekomme ich nicht wie eingestellt und damit erwartet, alle 60 s eine Datenaktualisierung, obwohl laut LED-Anzeige am Sender und laut Serial Monitor der Arduino IDE Daten gesendet werden.
Zunächst glaubte ich, dass die Aktualisierung nur sporadisch erfolgt, habe aber dann festgestellt, dass es immer nach ca. 20 Minuten passiert.
Mein Verdacht, dass es im Zusammenhang mit dem Senden der Flags nach jeder 20 Messung steht, hat sich insoweit bestätigt, da ich das Intervall probeweise von 20 auf 5 gesetzt habe und nun ca. alle 5 min. eine Datenaktualisierung bekomme. 
Das ist doch so sicher nicht gewollt?

TomMajor, 2020-04-23:
nein, das wäre nicht so gewollt.

Meine letzte Änderung ging auf einen Hinweis von @jp112sdl zurück
https://homematic-forum.de/forum/viewtopic.php?f=76&t=56395&start=140#p571712

ich habe diese Änderung selber noch nicht getestet da ich Jerome 100% vertraue in diesen Sachen :smile: 
oder ich habe noch einen Impl.fehler drin.

kann ein paar Tage dauern bis ich das selber testen kann

geht es denn mit dem alten Stand, also nur sendPeerEvent ohne if/else ?
https://github.com/TomMajor/SmartHome/commit/06ebe16f8cac40a22acb665e816420d0ae19175b

HMelzer, 2020-04-24:
Ich habe gerade die Programmdatei auf den alten Stand geändert und das Ganze nochmal getestet.
Es bleibt bei dem beschriebenen Phänomen, dass die Änderung auf dem Homematic Desktop immer erst bei der zwanzigsten Änderung auch angezeigt wird.

TomMajor, 2020-04-24:
habe es noch mal getestet, läuft alles einwandfrei, Werte werden jede Minute aktualisiert.
siehe screenshot
auch schön zu sehen das ACK nur in der 2. Nachricht (durch das BIDI flag)

aktueller HB-UNI-Sensor1 sketch von github
AddOn Ver. 2.04
RaspberryMatic Ver. 3.51.6.20200229
[image]

jp112sdl, 2020-04-25:
@HMelzer Ist es eine CCU2 ? Welcher Firmwarestand?

HMelzer, 2020-04-25:
Nein, est ist eine RaspberryMatic.
Davon habe ich zwei, eine mit Version 3.49.17.20191225
und eine zweite mit Version 3.49.17.20200131.
es geht aber bei beiden nicht, wie es eigentlich sollte.
Ändere ich die Zeile        if ((msgcnt % 20) == 2) {
                                         flags = BIDI | WKMEUP;
                                         }
wie dargestellt, funktioniert es natürlich, erfüllt aber nicht den Zweck.

Warum eigentlich ==2 und nicht ==1?

jp112sdl, 2020-04-25:
Ein Möglichkeit wäre noch, den RFD auf der RaspberryMatic mal mit Konsolen-Logging  zu starten und zu schauen, was ausgegeben wird, wenn ein Broadcast-Telegramm vom HB UNI Sensor1 kommt.

```
monit stop rfd

/bin/rfd -f /var/etc/rfd.conf -c -l 0

#Zum Beenden des RFD dann Strg+C drücken und wieder normal starten mit:
monit start rfd
```

HMelzer, 2020-04-25:
Danke!
Werde ich im Laufe des Tages ausprobieren und das Ergebnis mitteilen.

TomMajor, 2020-04-25:
und du sendest tatsächlich sich jede 60s ändernden Sensorwerte und nicht immer die gleichen dummy Werte? - nicht das du auf das "Letzte Änderung / Zeitstempel bleibt gleich" in der Zentrale reingefallen bist.

HMelzer, 2020-04-27:
Nachdem ich nun alle von mir getroffenen Änderungen zurückgenommen und gefühlt die Software 100 Mal neu installiert habe, bleibt es bei dem geschilderten Zustand.
Im Serial Monitor der Arduino IDE wird alle 60 sec. eine Messung durchgeführt und offenbar auch in Richtung Zentrale abgesetzt. Dort kommen die Werte aber immer nur nach ca. 22 Minuten an.
Auch das Konsolen Logging, wie von Jérôme vorgeschlagen, liefert das gleiche Ergebnis: Daten von diesem Gerät kommen immer erst nach ca. 22 Minuten. 
Im Moment muss ich erst einmal über das Problem weiter nachdenken und eventuell noch nach anderen Lösungen suchen. Für weitere Vorschläge bin ich offen. Werde mich zu gegebener Zeit wieder melden.

TomMajor, 2020-04-27:
sehr merkwürdig, dafür habe ich momentan keine Erklärung.
Sendest du sich ändernde Sensorwerte?

jp112sdl, 2020-04-27:
Und hast du mal den FreqTest laufen lassen?
Nicht dass es so ein _"geht gerade so"_ CC1101 ist

TomMajor, 2020-04-27:
das würde auch erklären warum es bei jedem 20. Mal bei BIDI geht, oder?
Weil er dann bis zu 6x sendet da er auf ACK wartet..

HMelzer, 2020-05-09:
Des Räsels Lösung ist, wie ich glaube, gefunden!
Ich hatte einige Arduino Pro Mini mit 5V und 16 MHz in meiner Sammlung und habe beim Aufbau des Sensors einen solchen erwischt, diesen aber so behandelt, als wäre es ein 3,3 V, 8 MHz-Typ.
Warum der sich dann so verhält, weiß ich zwar nicht, aber mit dem Original geht es natürlich genau so wie erwartet.

TomMajor, 2020-05-10:
ok, freut mich das es jetzt funktioniert.

closed 2020-05-10 by TomMajor