TomMajor/SmartHome #35: Brightness in CCU 999.99 Lux

JocB1, 2020-05-02:
Hello, 
I build the Uni Sens1 Sensor with the newest Asksinpp lib and the newest Addon for the CCU. In the darkness the CCU shows 999.00 Lux - may be an overrun problem... I use the TSL 2561 Sensor (and a BME280). 
Any Idea?

TomMajor, 2020-05-03:
you probably saw this:
https://github.com/TomMajor/SmartHome/tree/master/Info/SensorTest_Lux

I don't use the TSL2561 anymore because of problems in the sunlight, But it should be fine in darkness.

I would suggest to use debug output in the measure loop to check all values what's going on
https://github.com/TomMajor/SmartHome/blob/master/HB-UNI-Sensor1/Sensors/Sens_TSL2561.h#L90

JocB1, 2020-05-04:
Here my debug outputs: 

dark: 
BME280 Temperature x10  : 231
BME280 Pressure x10     : 9715
BME280 PressureNN x10   : 9715
BME280 Humidity         : 51
TSL2561 Sensitivity        : 0
TSL2561 Brightness Raw Full: 0  !!
TSL2561 Brightness Raw IR  : 0   !!
TSL2561 Brightness Lux x100: 99999  !!!!!!  --> CCU: 999.99 Lux 

indoor: 
BME280 Temperature x10  : 220
BME280 Pressure x10     : 9714
BME280 PressureNN x10   : 9714
BME280 Humidity         : 55
TSL2561 Sensitivity        : 2
TSL2561 Brightness Raw Full: 506
TSL2561 Brightness Raw IR  : 80
TSL2561 Brightness Lux x100: 20700   -> CCU: 207 Lux

TomMajor, 2020-05-05:
the overflow detection code for direct sunlight 2 years ago doesn't take zero raw values into account.
You could try to extend this first part of the if condition
(_brightnessRawFull == _brightnessRawIR)
with a logical AND
&& (_brightnessRawFull  != 0)
still not perfect for raw values where _brightnessRawFull is equal _brightnessRawIR
but have no other idea at the moment.

closed 2020-05-14 by TomMajor