TomMajor/SmartHome #36: HB-SEN-LJet mit Arduino Pro Mini

niels7418, 2020-05-31:
Hallo Tom, ich bin auf Dein Projekt Level Jet gestoßen und wollte es gleich nachbauen.
Ich habe es mit einen Arduino Pro Mini versucht. Das ganze habe ich nach Deinem Übersichtsschaltplan verdrahtet. Dein Skript habe ich leicht angepasst da ich eine Platine von Thomas verwende nach Jerome`s Skripten. Der Anlernbutton auf PIN 8 anstatt Pin 5 und die LED auf Pin 4 anstatt Pin 7. Nun zu meinem Problem, wenn ich die Platine an den Level Jet anschließe, blinkt jedes mal wenn der Level Jet misst die LED leicht. Daher gehe ich davon aus, dass der Arduino Daten empfängt. In der WebUi der Homematic bleiben alle Werte auf Null. Im Systemprotokoll der WebUi meldet er sich fleißig. Da ich ich die Schaltung 2 Mal aufgebaut habe, bin ich mir ziemlich sicher keinen Verdrahtungsfehler zu haben. Hast Du vielleicht einen Tipp woran es liegen könnte?

TomMajor, 2020-05-31:
Hmm, schwer zu sagen mit diesen Infos.
Wie genau sehen denn die Eintrage "im Systemprotokoll der WebUi" aus?
Wie gut kennst du dich mit Elektronik, serieller Schnittstelle usw. aus? Hilfreich wäre das Tx über RS232 auf einen extra Schnittstelle am PC zu legen, aber das Rx vom LevelJet zu lassen wie es ist. Dann z.B. schauen was diese Ausgabe sagt:
https://github.com/TomMajor/SmartHome/blob/master/HB-SEN-LJet/src/Leveljet.h#L123

niels7418, 2020-06-01:
Moin Tom, 
In der Elektronik bin ich eine Weile raus. Bin grade dabei mir die Sache wieder näher zu bringen. Da ich nicht wirklich Erfahrung mit SMD Bauteilen habe, wollte ich es mit einem Arduino probieren. 
Gestern hatte ich über den Sketch versucht den Arduino auszulesen. Da kam aber nur Kauderwelsch. Vielleicht habe ich auch einen Denkfehler in der Verdrahtung. In den Fotos kannst Du den log aus der CCU, die Ausgabe vom Arduino und wie ich die Verschaltung verdrahtet habe. 
Kann ich Leveljet direkt mit dem FTDI232 anschließen und dann über die Arduino Software auslesen? 

Danke für Deine Hilfe 

Gruß Niels

niels7418, 2020-06-01:
https://drive.google.com/drive/folders/1qZcd-UT4IzdZtcei6pygBHWbaEil7HN2?usp=sharing
Unter diesem Link sind die Fotos.

TomMajor, 2020-06-01:
da dies eher ein Aufbauproblem als ein Bug ist, bitte das Thema im Forum, z.B. hier fragen:
https://homematic-forum.de/forum/viewtopic.php?f=76&t=49512
Dort können ggf mehr Leute helfen und es ist nicht wirklich eine github issue.
Auf den Fotos kann man auch nichts erkennen, viel zu klein.

closed 2020-06-01 by TomMajor