TomMajor/SmartHome #39: Standardwerte für transmitDevTryMax, updIntervall, altitude usw.

HMelzer, 2020-06-04:
Hi Tom,

ich mühe mich schon seit geraumer Zeit erfolglos ab, zu ergründen, woher die Werte für die o.g. Daten kommen, die im Seriellen Monitor angezeigt werden:
[image]

Die Standardwerte, die ich im Sketch eingetragen habe, können es nicht sein:
[image]

Es können auch keine Werte sein, die aus der Homematik selbst kommen, da der Sensor noch nicht daran angemeldet ist.
Woher also bezieht das kompilierte Programm seine Werte bzw. wo wären diese ggf. zu ändern?

jp112sdl, 2020-06-04:
> Da hilft ein longpress des Config buttons, das macht ein komplettes device reset (lässt sich im seriellen Monitor beobachten)

... und dann werden die Werte mit denen in defaults() belegt.

HMelzer, 2020-06-04:
Genau das war es wohl!
Hat funktioniert, währe ich aber nie darauf gekommen.
Vielen Dank und ein schönes Wochenende.

closed 2020-06-04 by HMelzer