TomMajor/SmartHome #42: Version ist falsch

MathiasJZ, 2020-07-19:
habe gerade versucht, das hb-tm-devices-addon upzudaten.
nach dem alten Update bekomme ich immer noch die Meldung, es gäbe ein neues 2.5.1 und das ältere 2.5.0 wäre installiert.
Gruß,
Mathias

TomMajor, 2020-07-19:
Danke für den Hinweis. Habe gerade die Version auf 2.50 rückgesetzt.
Es wird eine neue Version geben, aber noch nicht alle Änderungen gemacht bzw. getestet,
Setzen der neuen Version war def. zu früh.

MathiasJZ, 2020-07-19:
Ich kann jetzt aber ich nenne sie mal 2.5.0a lassen, oder soll ich die 
originale 2.5.0 installieren?
Gruß,
Mathias

TomMajor, 2020-07-19:
Kannst alles so lassen wie es ist.
Es wurde nur die Textdatei mit der Version "zu früh" geändert, siehe commits.
Das eigentliche AddOn **hb-tm-devices-addon.tgz** wurde noch gar nicht angefasst.

TomMajor, 2020-08-02:
Seit gestern ist die neue Version draußen.

closed 2020-08-02 by TomMajor

MathiasJZ, 2020-08-02:
Gestern wurde sie auch installiert
Gruß, 
Mathias