TomMajor/SmartHome #43: hb-uni-sensor6 der UV-INDEX-Wert wird in der WEBUI nicht angezeigt

chianti2000, 2020-08-24:
Für den hb-uni-sensor6 mit VEML6075 wird der UV-Index Wert in der WEBUI nicht angezeigt. Das Feld UV-Index ist da.
Für den hb-uni-sensor5 mit VEML6070 funktioniert das.
Der serielle Monitor liefert UV-Werte Werte.
`AskSin++ V4.1.6 (Aug 24 2020 12:40:45)
MAX44009 found
SHT31 found
VEML6075 found
Sensor setup done
Serial: UNISENS061
Clock SYSCLOCK
Address Space: 32 - 79
CC init1
CC Version: 14
 - ready
tmBatteryLoad Voltage: 4395
Battery set low:  21
Battery set crit: 19
Config Changed: List0
ledMode: 1
lowBatLimit: 21
Battery set low:  21
transmitDevTryMax: 6
updCycle: 60
altitude: 0
SHT31 Temperature x10   : 327
SHT31 Humidity          : 38
MAX44009 Brightness x100: 6414337
VEML6075 UVA            : 414
VEML6075 UVB            : 1325
VEML6075 UVI            : 2.0
<- 17 01 84 70 A5A561 000000 01 47 2A 80 26 00 61 E0 01 00 11 2B 00 14  - 534
SHT31 Temperature x10   : 332
SHT31 Humidity          : 37
MAX44009 Brightness x100: 6377473
VEML6075 UVA            : 413
VEML6075 UVB            : 1325
VEML6075 UVI            : 2.0
<- 17 02 A2 70 A5A561 371AA8 01 4C 2A 80 25 00 61 50 01 00 11 2B 00 14  - 794
-> 0A 02 80 02 371AA8 A5A561 00  - 921
waitAck: 01
SHT31 Temperature x10   : 337
SHT31 Humidity          : 37
MAX44009 Brightness x100: 6377473
VEML6075 UVA            : 416
VEML6075 UVB            : 1321
VEML6075 UVI            : 2.0
`
viele Grüße
Gerd

TomMajor, 2020-08-24:
Hmm, kann das gerade aus Zeitgründen nicht selbst testen, aber die message sieht gut aus.

bin nicht sicher ob es daran liegt, aber ändere bitte mal die Zeile 127
`<logical type="integer" min="0" max="12" />`
in
`<logical type="integer" min="0" max="120" />`
im 
hb-uni-sensor6.xml

also über SSH editieren, in 
/usr/local/addons/hb-tm-devices-addon/firmware/rftypes

Danach, bei xml Änderungen immer den Sensor ablernen/löschen und neu anlernen.
Hoffe das hilft.

chianti2000, 2020-08-25:
Danke für die Hilfe, die Änderung hat leider nichts gebracht. Beim hb-uni-sensor5 mit VEML6070 funktioniert die Anzeige perfekt.
Wenn's keine Lösung gibt werde ich erstmal mit dem hb-uni-sensor5 mit VEML6070 leben müssen.

TomMajor, 2020-08-25:
Muss gehen, ist sicher nur eine Kleinigkeit.
Ich sehe ja den UV-Index 2.0 in deiner message an die Zentrale (das letze Byte, 14hex).

mach mal bitte einen screenshot von diesem Gerät im WebUI, da wo die Werte zu sehen sind.

Device Model in Arduino sketch und im hb-uni-sensor6.xml stimmen ganz sicher überein?

chianti2000, 2020-08-26:
Device Model und  hb-uni-sensor6.xml stimmen überein. Hier der Screenshot zum letzten Messwert im seriellen Monitor.
[image]
`AskSin++ V4.1.6 (Aug 24 2020 12:40:45)
MAX44009 found
SHT31 found
VEML6075 found
Sensor setup done
Serial: UNISENS061
Clock SYSCLOCK
Address Space: 32 - 79
CC init1
CC Version: 14
 - ready
tmBatteryLoad Voltage: 4395
Battery set low:  21
Battery set crit: 19
Config Changed: List0
ledMode: 1
lowBatLimit: 21
Battery set low:  21
transmitDevTryMax: 6
updCycle: 60
altitude: 117
SHT31 Temperature x10   : 268
SHT31 Humidity          : 45
MAX44009 Brightness x100: 12976120
VEML6075 UVA            : 1042
VEML6075 UVB            : 1725
VEML6075 UVI            : 3.0
<- 17 01 84 70 A5A561 000000 01 0C 2A 80 2D 00 C5 FF F8 00 11 2B 00 1D  - 536
SHT31 Temperature x10   : 266
SHT31 Humidity          : 45
MAX44009 Brightness x100: 12828664
VEML6075 UVA            : 1046
VEML6075 UVB            : 1721
VEML6075 UVI            : 3.0
<- 17 02 A2 70 A5A561 371AA8 01 0A 2A 80 2D 00 C3 BF F8 00 11 2B 00 1D  - 796
-> 0A 02 80 02 371AA8 A5A561 00  - 921
waitAck: 01
SHT31 Temperature x10   : 269
SHT31 Humidity          : 44
MAX44009 Brightness x100: 12754936
VEML6075 UVA            : 1045
VEML6075 UVB            : 1713
VEML6075 UVI            : 3.0
<- 17 03 84 70 A5A561 000000 01 0D 2A 80 2C 00 C2 9F F8 00 11 2B 00 1D  - 1667
`

Hier zum Vergleich das Ganze für den hb-uni-sensor5 mit VEML6070:
[image]

`
AskSin++ V4.1.6 (Aug  3 2020 19:38:11)
MAX44009 found
SHT31 found
VEML6070 found
Sensor setup done
Serial: UNISENS051
Clock SYSCLOCK
Address Space: 32 - 79
CC init1
CC Version: 14
 - ready
tmBatteryLoad Voltage: 4395
Battery set low:  21
Battery set crit: 19
Config Changed: List0
ledMode: 1
lowBatLimit: 21
Battery set low:  21
transmitDevTryMax: 6
updCycle: 60
altitude: 115
SHT31 Temperature x10   : 258
SHT31 Humidity          : 44
MAX44009 Brightness x100: 7778305
VEML6070 UV raw value   : 481
VEML6070 UV-Index       : 2
<- 17 01 84 70 A5A551 000000 01 02 2A 80 2C 00 76 B0 01 00 11 2B 00 02  - 507
`

TomMajor, 2020-08-26:
ok, ich versuche das mal die nächsten Tage selber zu testen.
Die Daten sehen soweit alle gut aus. Kann eigentlich nur eine Kleinigkeit im xml oder so sein.

chianti2000, 2020-08-26:
Danke erstmal 👌

TomMajor, 2020-08-29:
konnte den Fehler nachstellen und auch fixen. Mache morgen eine neue AddOn Version.

chianti2000, 2020-08-30:
Vielen Dank Tom.

TomMajor, 2020-08-31:
Hi Gerd,
so, die neue Version ist oben. ich empfehle folgendes Vorgehen:

1. alle vorhandenen HB-UNI-Sensor6 aus der Zentrale deinstallieren/löschen
2. HB-TM-Devices-AddOn V2.52 installieren (Zentrale wird dabei neu gestartet)
3. HB-UNI-Sensor6 mit VEML6075 neu anlernen

dann geht hoffentlich auch der UV-Index für den VEML6075  (bei mir funktioniert es)

chianti2000, 2020-08-31:
Ja, kann bestätigen, dass der UV-Index jetzt angezeigt wird. Es gibt zwar einen Offset von -0,1 von der WEBUI zum seriellen Monitor (immer konstant -0,1 bei verschiedenen UV-Werten) aber das kann ich vernachlässigen und fällt unter Messfehler 😉.
Vielen Dank nochmals für deine Arbeit.
[image]

closed 2020-08-31 by chianti2000

TomMajor, 2020-09-01:
ja das Runden war nicht optimal, ich habe das gerade geändert:
https://github.com/TomMajor/SmartHome/commit/86f4feb4c0b5dc7f935a1b4ee87d7401c3b25ee2

durch die Änderung in Zeile 83 wird der UVI im seriellen Monitor mit 2 Nachkommastellen geschrieben.

Und durch die Änderung in Zeile 74 sollte der UVI jetzt für das WebUI korrekt gerundet werden. z.B. ein Rohwert von 2.04 ergibt 2.0 im WebUI, ein Rohwert von 2.06 ergibt 2.1 im WebUI.

[image]