TomMajor/SmartHome #46: Where to buy?

andispammonyer, 2021-06-09:
Hi Tom,

ich würde dir gerne drei Helligkeitssensoren abkaufen.
Mein Fokus liegt rein auf der Detektion der Sonneneinstrahlung. 
Möchte über ein Jahr evaluieren ob das Flachdach von meiner Garage sich als PV Anlage rechnen würde. Dazu muss ich wissen wieviel Sonne ich dort zu erwarten habe.

TomMajor, 2021-06-09:
Sorry, verkaufen tue ich hier nichts. Und das ist auch kein Thema für eine github issue.
Zum Thema Sonnensensor magst du vlt. mal einen Blick hier draufwerfen:
https://homematic-forum.de/forum/viewtopic.php?f=76&t=65391

andispammonyer, 2021-06-10:
Würde es eine Mailfunktion bei Github geben, hätte ich dir darüber geschrieben.
Alles klar dann schau ich anderweitig weiter.
Danke

closed 2021-06-10 by andispammonyer

TomMajor, 2021-06-10:
kein Problem, ist nur ungewöhnlich diese Art von Anfrage in den Issues. 😸 

Eine gute Anlaufstelle für so was wären m.E. im HM Forum die Unterforen
"Hardwareentwicklung und Selbstbau von Aktoren und Sensoren"
und/oder
"Markt".

Außerdem würde ich dann an deiner Stelle besser spezifizieren ob du z.B. einen Helligkeitssensor MAX44009 suchst oder jemand der dir Komplettgeräte aufbaut.