TomMajor/SmartHome #47: HB-UNISERSOR peering mit Thermostat in CCU: Zeitslotanpassung

SonOfAbaddon, 2022-11-29:
Hi Tom,

Ich habe die HB-UNI nun schon lange laufen und vor ein paar Monaten von FHEM HM_CUL zu RaspberryMAtic/HMCCU umgestellt. Leider treffen die HB-UNI im direkten peering den Zeitslot der HM-CC-RT-DN so gut wie nie. Meine Nachforschung hat ergeben, dass mit dem bei mir verbauten 8KHz Quarz der Sync zum Thermostat eher glückssache ist. auf HB-UNISENSOR2 habe ich den Sketch gewechselt. Ich möchte daher gerne auf 32,XXKHz Quarze umsteigen, damit das zuverlässig(er) läuft und meine Thermostate trotz Offset nicht ständig überheizen.

Folgende Punkte sind für mich offen:
- welchen 32,768KHz Quarz kannst du empfehlen und müssen dann Pufferkondensatoren (4p?) zusätzlich angebracht werden?
- Muss der Bootloader mit anderen Fuses neu geflasht werden oder reicht es im Sketch auf RTC zu schwenken?
- Braucht der Sketch weitere Anpassungen?

Vorab vielen Dank für eine Antwort!

TomMajor, 2022-11-29:
Hi,

-  Thema Quarz und load capacitance, siehe hier
https://forum.fhem.de/index.php/topic,57486.msg797983.html#msg797983
- bootldr nein, es reicht im Sketch das RTC define zu aktiveren
- dein Problem mit dem Zeitslot hatten schon viele Leute, kein gutes Design von eQ-3 IMHO, papa hatte dafür auch die Lib angepasst, siehe hier einige Posts zu dem Thema
https://forum.fhem.de/index.php/topic,20620.msg965198.html#msg965198

SonOfAbaddon, 2022-12-06:
Hi,

so, die Quarze sind bestellt und werden dann die bisherigen externen 8KHz Quarze ersetzen. Reichelt hatte sogar welche mit 6pF Lastkapazität im Portfolio, so dass keine weiteren C nötig sind.

Mal sehen, ob ich mit papas angepasster Lib das auch in den Griff bekomme. ISt extrem nervig, dass die Heizung tagsüber übersteuert und Nachts zu schnell abfällt und dann auch unnötig zwischenheizt.

SonOfAbaddon, 2022-12-10:
Ich habe 8 meiner 9 HB-UNI umflashen können, der Sync zu 3 Thermostaten sieht erst einmal gut aus. Die letzte Platine macht nach dem Flashen allerdings Probleme, ich bekomme ohne Quarz am XTAL keinen sync in der Arduino ISE hin. Habe stundenlang diverse Fuse- & Loader-Kombis ausprobiert.
Jetzt ist der 8MHz Quart wieder am XTAL, der ATmegaBOOT_168_atmega328_pro_8MHz.hex loader ist drauf und die Fuses auf -U lfuse:w:0xFF:m -U hfuse:w:0xD2:m -U efuse:w:0xFF:m gestellt. Allerdings wirft der Sensor nun nach dem Booten folgendes in die Konsole und eine Kopplung an Raspberrymatic ist nicht mehr möglich:
19:21:12.135 -> AskSin++ v5.0.2 (Dec 10 2022 19:20:16)
19:21:12.135 -> BME280 found
19:21:12.135 -> Sensor setup done
19:21:12.135 -> Serial: UNISENS13D
19:21:12.135 -> Clock SYSCLOCK
19:21:12.135 -> Address Space: 32 - 85
19:21:12.135 -> CC init1
19:21:12.135 -> CC Version: 14
19:21:12.135 ->  - ready
19:21:12.368 -> tmBatteryLoad Voltage: 3089
19:21:12.368 -> Battery set low:  21
19:21:12.415 -> Battery set crit: 19
19:21:12.415 -> Config Changed: List0
19:21:12.415 -> ledMode: 1
19:21:12.415 -> lowBatLimit: 21
19:21:12.415 -> Battery set low:  21
19:21:12.415 -> transmitDevTryMax: 6
19:21:12.415 -> updCycle: 600
19:21:12.415 -> altitude: 0
19:21:12.415 -> tempOffset x10: 0
19:21:12.415 -> presOffset x10: 0
19:21:12.415 -> humiOffset x10: 0
19:21:17.623 -> BME280 Temperature x10  : 197
19:21:17.670 -> BME280 Pressure x10     : 10043
19:21:17.670 -> BME280 PressureNN x10   : 10043
19:21:17.670 -> BME280 Humidity x10     : 494
19:21:17.670 -> AbsHumidity x100        : 877
19:21:17.670 -> DewPoint x10            : 94
19:21:17.670 -> <- 19 01 84 70 000107 000000 00 C6 27 3B 33 00 00 00 00 01 0C 11 03 6D 00 5E  - 407
19:21:51.918 ->  debounce
19:21:52.010 ->  pressed
19:21:52.102 ->  released
19:21:52.149 -> <- 1A 02 84 00 000107 000000 15 F1 12 55 4E 49 53 45 4E 53 31 33 44 70 01 01 01  - 475
19:21:52.149 -> 
19:21:57.502 -> AskSin++ v5.0.2 (Dec 10 2022 19:20:16)
19:21:57.502 -> BME280 found
19:21:57.502 -> Sensor setup done
19:21:57.502 -> Serial: UNISENS13D
19:21:57.502 -> Clock SYSCLOCK
19:21:57.550 -> Address Space: 32 - 85
19:21:57.550 -> CC init1
19:21:57.550 -> CC Version: 14
19:21:57.550 ->  - ready
19:21:57.783 -> tmBatteryLoad Voltage: 3085
19:21:57.783 -> Battery set low:  21
19:21:57.783 -> Battery set crit: 19
19:21:57.783 -> Config Changed: List0
19:21:57.783 -> ledMode: 1
19:21:57.783 -> lowBatLimit: 21
19:21:57.783 -> Battery set low:  21
19:21:57.783 -> transmitDevTryMax: 6
19:21:57.783 -> updCycle: 600
19:21:57.783 -> altitude: 0
19:21:57.783 -> tempOffset x10: 0
19:21:57.783 -> presOffset x10: 0
19:21:57.783 -> humiOffset x10: 0
19:22:03.018 -> BME280 Temperature x10  : 196
19:22:03.018 -> BME280 Pressure x10     : 10042
19:22:03.065 -> BME280 PressureNN x10   : 10042
19:22:03.065 -> BME280 Humidity x10     : 514
19:22:03.065 -> AbsHumidity x100        : 907
19:22:03.065 -> DewPoint x10            : 99
19:22:03.112 -> <- 19 01 84 70 000107 000000 00 C5 27 3A 36 00 00 00 00 01 0C 0D 03 8B 00 63  - 407

Was sagt die letzte Zeile aus? Kommt im normalen Sensorintervall vor.

Habe ich mir den Atmega irgenwie zerlegt? denn ohne den 8MHz am XTAL mit den Fuses U lfuse:w:0xE2:m -U hfuse:w:0xD2:m -U efuse:w:0xFF:m für den 8MHz Internen Oszi komme ich zar mit AVRdude zum Setzen der Fuses und kann alles auslesen, aber nur mit 19200baud, die ISE wie oben beschrieben bekommt mit den normalen 57600er rate keine Verbindung.

Danke!

TomMajor, 2022-12-12:
bitte solche user-spezifischen Probleme im HM Forum/Hardwareentwicklung und Selbstbau
https://homematic-forum.de/forum/viewforum.php?f=76
anfragen, eine github issue ist dafür der falsche Platz.

closed 2022-12-13 by SonOfAbaddon